dchristl/esp32_nat_router_extended #3: esp rebooting

gjroots, 2022-01-07:
**device model : esp32-wroom**

**firmware version :  esp32nat_v.1.0.1.zip (2nd release)**

description :
        Hai  developer ,  i flash latest release into my esp32  using esp32flashtool.exe  , after flashing done  it continuously rebooting .. not showing  "ESP32_NAT_Router"  signal   .. 
  i check in putty ... it continuously rebooting .... 
*error log version*  : [errorlog.txt](https://github.com/dchristl/esp32_nat_router_extended/files/7826923/errorlog.txt)
.........

**firmware version :  esp32nat_v.1.0.0_beta.zip (1st release)**
 *so that i tested beta version esp32nat_v.1.0.0_beta.zip it works fine but during scan wifi it reboots .... and after not show any wifi "ESP32_NAT_Router" signal ... it say error ........
then i pressed reset button then it show signal until it stay in idle no wifi signal showing... after no list of ssd showing....
[image]

please make new version and fix this problem ...

dchristl, 2022-01-09:
Hello gjroots,

your 1st problem seems to be a memory or flash issue. Have you installed any other firmware before? Can you try to erase the whole flash before updating to the new version:

` esptool.py erase-flash` 

I know about your second problem, but don't have currently enough time to solve this. The device dsiconnects after changing the Wifi-Mode and reboots. Sorry for the inconvenience, but stay tuned until I will resolve this issue.

gjroots, 2022-01-14:
@dchristl  
hai...

ok i can under stand ..
i am  also working on same project based on his  [paoloinverse](https://github.com/paoloinverse/esp32_nat_router_pseudo-mesh)   code . I successfully designed "scan" functionality it fully working. .. but minor bugs are present .. under  testing .... if it complete i share my project ... 

I notice that  
 **If you want to enter non-ASCII or special characters (incl. ' ') you can use HTTP-style hex encoding (e.g. "My%20AccessPoint" results in a string "My AccessPoint").** 

^^ this is not required  ..... while form submission    just add this
`<form action='' method='GET'  enctype="application/x-www-form-urlencoded">`

`enctype="application/x-www-form-urlencoded`    ...
this will auto set spaces and special characters  with their equivalent    .... it works fine for me

Please can you share documentation  of your project how to setup and build .  during building i am getting error ... i cant understand..

dchristl, 2022-01-20:
Hello gjroots,

I deployed a new Beta-Release, adressing your issues with the scanning and the latest release.

 @gjroots 
> I notice that
If you want to enter non-ASCII or special characters (incl. ' ') you can use HTTP-style hex encoding (e.g. "My%20AccessPoint" results in a string "My AccessPoint").

I don't know what you mean, this is the documentation of the cloned project and corresponds to the nvs (CLI) - functionality. I replaced every request in the old project with POST and forms. There is no encoding or tags needed.

The documentation is coming soon ;)

dchristl, 2022-01-23:
This is resolved with [1.1.0-release](/dchristl/esp32_nat_router_extended/releases/tag/v1.1.0)

The missing documentation will be handled at #10

closed 2022-01-23 by dchristl

gjroots, 2022-01-23:
thank you............ @dchristl