dchristl/esp32_nat_router_extended #4: Allways reboot

pipi61, 2022-01-22:


dchristl, 2022-01-22:
Hi pipi61,

it seems to be connected to issue  #3. I already resolved this, so please can you try the latest beta available [here](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v1.0.2.beta)

If this resolve your issue, please give feedback.

Thank you

dchristl, 2022-01-23:
This is resolved with [1.1.0-release](/dchristl/esp32_nat_router_extended/releases/tag/v1.1.0)

closed 2022-01-23 by dchristl

pipi61, 2022-01-23:
Thank You for fast reply. The Beta version good work. I measure 10Mbps transfer speed, this good.
Good work.
[image]
one note: wifi scanning only works for the first time. After the ssid and pass settings, the wifi scan did not find any station, but the router is still working

pipi61, 2022-01-27:
Hi!
in  v1.1.1 the scan working.
Thank You
comment: I found a lost letter "v" :))
[image]

pipi61, 2022-01-29:
idea: I use it in another project, wifi.scannetwork runs immediately after poweron (before starting the router parameters) and stores the wifi host parameters in variables. If it is in the request station list of the web page, no restart and reconnection is required, only a list from variables

dchristl, 2022-01-29:
Thanks for your feedback,

> comment: I found a lost letter "v" :))

This should be solved with the latest [Beta](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v1.2.0-beta)

> idea: I use it in another project, wifi.scannetwork runs immediately after poweron (before starting the router parameters) and stores the wifi host parameters in variables. If it is in the request station list of the web page, no restart and reconnection is required, only a list from variables

Sorry, but I didn't really understand your mentioned workflow. Do you scan every time you reboot the controller? This takes up to 20s and this is too long for a function you use not so much. Otherwise you have to set a flag (user wishes a scan) and also have to trigger a reboot/ reconnect. 
In my case the reboot can be replaced with a reset of the connection and a reconnect. But this is cumbersome and takes as much time as a reboot. My workflow is like:

1. disconnect from AP wifi
2. disconnect all clients
3. scan and save the result in nvs
4. Reboot
5. Show scan result on next client request 
6. Delete variable 

Regards