dchristl/esp32_nat_router_extended #23: Reset pin

ErfanDL, 2022-02-24:
Hi. I love this project very much. is it possible to add an option for factory reset device by hardware pin ?

thanks.

dchristl, 2022-03-19:
Hello ErfanDL,

you can [try the new 2.0.0 beta release](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v2.0.0_beta) and read the [documentation](https://github.com/dchristl/esp32_nat_router_extended/blob/dev/docs/advanced.md#resetting-the-device-erasing-the-flash) how to use this feature. Any feedback and test will be appreciated. Thanks.

ErfanDL, 2022-03-20:
> Hello ErfanDL,
>
> you can [try the new 2.0.0 beta release](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v2.0.0_beta) and read the [...]

Hi.
Nice work. Thank you for your attention.

commit 6ab58603 (2022-03-19): Updated documentation, closes #23
closed 2022-03-25 by dchristl