dchristl/esp32_nat_router_extended #32: v2.2.0 constant reboot

blueskiesok, 2022-06-08:
on older version 1.0.2beta works.
when trying v2.2.0 constant reboot similar to previous reboot issue reported.

dchristl, 2022-06-08:
The partition table has changed at the [v2.1.0 ](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v2.1.0). An update is not possible, you have to execute a full reset like described [here](https://github.com/dchristl/esp32_nat_router_extended#first-install-reset)

blueskiesok, 2022-06-09:
This was on a fresh full install?

dchristl, 2022-06-10:
Have you erased the flash before? If not, please try. Otherwise I need more details about your device and some kind of log (reboot and debug flash log). I only have 3 different devices for testing, so maybe this is an other one with other specifications.

blueskiesok, 2022-06-12:
Just to check, I tried using Linux instead of windows.  Everything flashed and worked.  So apparently my problem has something to do with Windows 10.

dchristl, 2022-06-13:
Great to hear that, maybe your esp-idf on Windows is too old. Some older versions had some problems with custom partition tables.

closed 2022-06-13 by dchristl