dchristl/esp32_nat_router_extended #35: Range extender

Roberto-spm, 2022-08-12:
Hello, congratulation for excellent work!
I'm trying to set up a range extender with esp32 and your compiled firmware.
Basically everything works, and I was able to give internet line to  devices.
My problem is that the extender has a different netmask, so the devices connected to extender and the ones connected to main router can not reach each other.
I  would like to create an homogeneous network
Is it possible? Am I missing some setting?

Best regards
Roberto

dchristl, 2022-08-13:
Hello Roberto,

the change of the netmask is currently not possible. I will try to add this feature in the next release, but currently, I sadly have no time. ;)

Best regards,
Danny

dchristl, 2022-09-03:
Hello Roberto,

I've added the option for changing the netmask in the [current Beta-Release](https://github.com/dchristl/esp32_nat_router_extended/releases/tag/v.3.0.0.beta). I have no option test  it with a Class A or B network. Could you please check it with your setup and give me feedback? 

Thanks in advance,
Danny

commit 899b5525 (2022-09-19): closes #35
added new picture for documentation