dchristl/esp32_nat_router_extended #40: Is the second core used?

mariusl, 2022-10-09:
Not an issue but some advise please. I was wondering if you used the second core at all. I briefly scanned and found the led_status thread but I am not sure where it runs.
The reason I am asking is that I would like to run extra code on the second core (if available) to talk to a RF95 radio and scan some inputs via the ADC. Some information will be displayed on a simple webpage.

dchristl, 2022-10-09:
Hello mariusl,

I hope I understand you correctly. There are several threads (3 at max, depending on the config) involved in the whole system, but no CPU pinning is used. FreeRTOS is a single processor system by default, therefore without CPU pinning only one core is used. Do you see some different behaviour? Imho your project with the second core should be possible without issues. 

Best Regards,
Danny

mariusl, 2022-10-09:
Hi Danny,
I have not done any work on the project yet so I cannot say that it behaves in any way. I was under the impression that one can specify where your code must run. I also thought that goes for using RTOS. It would seem I did not read enough on the matter.
I believe that all WiFi code runs on the first core by default and cannot be moved but one can specify user code or segments of code to run on the other core. I just wonder if that is true for RTOS though. I was under the impression (mistakenly probably) that it was possible.
I will go read again. Thanks for the insight.

mariusl, 2022-10-09:
I did read a bit more and found this link [RTOS and dual core tutorial](https://microcontrollerslab.com/esp32-dual-core-freertos-arduino-ide/#:~:text=The%20ESP32%20development%20board%20consists,is%20used%20for%20RF%20communication.)

From this it seems that all application code runs in the second core and all RF stuff runs on the first core. The question would be how much of your code runs on core-1 and how much can I run on core-2 without screwing up things to much. Also I don't know if RTOS is running any of the WiFi stuff or webserver etc.

dchristl, 2022-10-10:
I've played a little bit around with my ESP32s, but none of them behaves like mentioned in your link. If I print out the used core in the main application and the threads ther are all running on CPU0. 
I would suggest you implement your new functionality either at the second CPU or as threads. like I do. Imho both scenarios will fulfill what you are trying to achieve.

mariusl, 2022-10-10:
Hi Danny,
Tanks for the effort man. If your threads report running on CPU0 then I expect that your whole app is running on that. So I agree with you then my stuff can easily live on the second CPU.

While I have your attention, what will be the best way to get data out to an IP from any sub function. Like using MQTT that is received by the LoRa network. I need to go in the back door to send data to a specific IP. How would you approach that?

dchristl, 2022-10-10:
I'm not really I microcontroller expert, but I think the easiest way is to call the IP of the target system directly, instead of using MQTT. But this depends on the target system, is there already an MQTT-Client in use than the queue will be the best way, other wise a simple REST webservice will be easier to implement.

mariusl, 2022-10-10:
Thanks for the help I do appreciate it very much.

closed 2022-10-10 by mariusl