dchristl/esp32_nat_router_extended #41: undefined reference to `app_main'

Hasaken, 2022-10-23:
Hi.

I'm trying to build on "esp-idf 4.3.2" and get following error in the end.

[972/973] Linking CXX executable esp32_nat_router.elf
FAILED: esp32_nat_router.elf 
cmd.exe /C "cd . && C:\Users\Professional\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\esp32_nat_router.elf.rsp -o esp32_nat_router.elf  && cd ."
c:/users/professional/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x14): undefined reference to `app_main'
c:/users/professional/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
C:/Users/Professional/.espressif/frameworks/esp-idf-v4.3.2/components/freertos/port/port_common.c:129: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Could you please advise me how to avoid this error?

dchristl, 2022-10-23:
Hi Hasaken,
nearly impossible to tell what is wrong here. Your setup seems broken and the error is really generic. I just wonder why there is a cpp-compiler (g++) involved, this project is c-code only. Have you seen [build instructions](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md) in the docs?

closed 2022-10-28 by dchristl

jonzobot, 2023-03-27:
Just commenting that trying to compile on linux using idf v. 5.0 I am running into the same compilation error. Hasaken, did you figure out the issue?

Hasaken, 2023-03-29:
Hi guys.

@jonzobot It was C language environment problem. You should check Platformio settings or config files inside .vscode folder.