dchristl/esp32_nat_router_extended #46: AP power management

sgrizzi, 2022-11-24:
Hi, I have been using this wifi extender for a while now. It is just great, very easy to use and very effective!
I just noticed that, since the AP is operating on the same channel as the incoming wifi signal from the router, it tends to "disturb" nearby devices which are themselves directly connected to the same router (e.g. my wireless printer).
Is there any possibility to reduce the AP output power level ? ideally via the webinterface, but also a command line would do.... or a modification to a config file...
Best regards.

dchristl, 2022-11-24:
Hello sgrizzi,
thanks yor your kind feedback. 
> Is there any possibility to reduce the AP output power level ? 

Sadly there is no such option neither in the Nat router nor the ESP32 itself. So it is also not possible to implement. But I plan to add a feature to change the AP channel in the config. This may help you for your problem.

Kind Regards,
Danny

dchristl, 2022-11-24:
Sorry, have to improve myself. As pointed out in the [official documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/network/esp_wifi.html?highlight=wifi#_CPPv419esp_wifi_set_config16wifi_interface_tP13wifi_config_t)  it is not possible to change the channel in soft-AP+station mode:

> 3. ESP32 is limited to only one channel, so when in the soft-AP+station mode, the soft-AP will adjust its channel automatically to be the same as the channel of the ESP32 station.

So this is a ESP32 limitation to use only one channel for softAP and STA.

sgrizzi, 2022-11-24:
Hi Danny, thanks for your prompt feedback. Unfortunately this was also my fear / having just "one" radio it would have been difficult to operate on two channels... hence my question on power management.
Clear, we will need to live with this setup if no power management is possible.
Thanks a lot!

closed 2022-11-24 by sgrizzi