dchristl/esp32_nat_router_extended #47: Communication link tester

sgrizzi, 2022-11-24:
The esp32 extender does turn on very fast, so I have been using it to simulate a "poor wifi" condition and verify how resilient my application code was.
Plugging and unplugging the usb cable of the demoboard is a bit cumbersome, and I was wondering if it was possible, for example, to use the on board "boot" button to toggle the AP on or off. Is there any possibility to do so?

My second question would be a feature extension to enable the extender to become a "communication link tester".
For example, via wifi I connect my application board to the extender.
Via the extender, my board then connects to a remote server.
Would it be possible to "cut" the communication to server, on command, without deactivating the wifi connection of my board to the extender, so that I could simulate a "server down" situation while the wifi connection is still active?

Best regards, Gabriele

dchristl, 2022-11-24:
> Plugging and unplugging the usb cable of the demoboard is a bit cumbersome, and I was wondering if it was possible, for example, to use the on board "boot" button to toggle the AP on or off. Is there any possibility to do so?

On all my dev-boards you can hold the EN-Button, that disables the ESP32 and no WiFi-connection is available. If you release it the board reboots and reconnects. Maybe this work for your board as well? 

> Would it be possible to "cut" the communication to server, on command, without deactivating the wifi connection of my board to the extender, so that I could simulate a "server down" situation while the wifi connection is still active?

If I understand you correctly you will simulate a disconnection between a remote server and the ESP32, but the soft-AP client stays connected? There a lot of (easier) ways to achieve this than to add such a feature in the NAT router. You can:

- shutdown the remote server if it is under your control
- Disable the internet temporarily for the ESP32 in your router
- Add a invalid DNS entry for this server in your router
- Add an entry to your firewall for the remote server 
- Change your code temporarily to an invalid server for testing purposes
- Change the DNS server in the NAT router's advanced config to something invalid.

sgrizzi, 2022-11-24:
> On all my dev-boards you can hold the EN-Button, that disables the ESP32 and no WiFi-connection is available. If you release it the board reboots and reconnects. Maybe this work for your board as well?
I will try and see if this works. A toggle function would be handy so that you can do some other work while wifi is off/on....

> If I understand you correctly you will simulate a disconnection between a remote server and the ESP32, but the soft-AP client stays connected
Yes, indeed, this is my purpose.
Unfortunately I have no control on the remote server, nor can I switch off my router as I need internet connection for other reasons (also to the server in question).
I would just need to switch of the server connection "just" to my board and the esp32 expander would be a nice switch along the communication path, and ensure that the board stands the disconnection and a subsequent reconnection.

dchristl, 2022-11-24:
> I will try and see if this works. A toggle function would be handy so that you can do some other work while wifi is off/on....

This is a kind of toggle function. You just have to bridge EN and GND with a jumper wire to achieve this. Its easy and hardware based, so no serial connection required.

> Unfortunately I have no control on the remote server, nor can I switch off my router as I need internet connection for other reasons (also to the server in question).

What about my several other suggestions? If I had such a reuirement, I would simply disable internet (not WifI, only internet) just for my board in my router. The ESP32 stays connected, but can't connect to any server in the internet. This works instant and can be switched in seconds. And you also don't need any serial connection to the board. 

Hope, this helps to solve your problems.

sgrizzi, 2022-11-24:
Hi Danny, thanks for your support.
Yes, the EN to GND solution does work! And it is indeed good enough. Thanks.

Disabling internet access for the esp32 expander on the router also works. It is not very simple... but, again, good enough for my purpose.
Thanks again for your suggestions.
Best regards, Gabriele

closed 2022-11-24 by sgrizzi