dchristl/esp32_nat_router_extended #49: Bridge

JuandreG, 2022-12-09:
Hi.

Awesome enhancements to @martin-ger's project.

Can this be used as a bridge? It seems that martin's cannot be in "bridge mode" and I was hoping this might have that feature?

dchristl, 2022-12-12:
Hi JuandreG,

thanks for yur kind feedback. I'm not really sure what you mean by bridge mode. Do you mean layer2 bridging? This is only possible with ethernet/LAN, like described [here](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap). 
Bridging with WiFi is not suported by ESP32 and AP/STA mode. For further details, you can have a look into [this thread](https://www.esp32.com/viewtopic.php?t=17324)
I don't know what you are trying to achieve, but you can set the same SSID and password for your AP-network like the STA-network. Your devices will connect to the strongest signal autmatically. 

Regards,
Danny

JuandreG, 2022-12-12:
Thanks for responding.

When logging in to a wifi router, in some cases you have the option of setting it in "bridge mode" this is what I mean.

So the problem I am trying to work out is when I use your perfectly functioning project for repeating wifi the clients of that repeater would get an IP address of lets say 192.168.4.6. So the problem is I can't then ping that client and I am also not able to post data to that client (ESP32) with a small webserver running. I did get to posting to the client using port forwarding but I'd rather use the IP of the client for posting. 

When I set a static IP for the repeater, lets say 10.96.253.1, I am then able to ping the client device, 10.96.253.4, with but I am still not able to post data to the client.

Any advice in this regard?

dchristl, 2022-12-12:
> When logging in to a wifi router, in some cases you have the option of setting it in "bridge mode" this is what I mean.

This is what I meant with layer 2 bridging. This sadly isn't possible with the ESP32 (see linked thread in 1st answer)

> So the problem I am trying to work out is when I use your perfectly functioning project for repeating wifi the clients of that repeater would get an IP address of lets say 192.168.4.6. So the [...]

> When I set a static IP for the repeater, lets say 10.96.253.1, I am then able to ping the client device, 10.96.253.4, with but I am still not able to post data to the client.

I think the problem is the different network classes. The default of the router is Class C (192.168.XX) and your repeater uses Class A (10.0.XXX). The connected devices can only communicate with each other if the same networkmask is used. 
I've implemented an option for this in v3.0.0. If you have this or a newer release you can use the netmask override function (advanced config) to adapt the netmask of your upstream (it seems like Class A) to the ESP32. Be aware that the IP of ESP32's web frontend will change! 

Let me know if this solves your problem.

JuandreG, 2022-12-12:
Thank you so much for your advice.

I will comment here when I tried it out and close if it worked.

JuandreG, 2022-12-13:
Hi again.

I tried what you suggested but sadly it didn't work.

I was not able to post to the device or post from the device.

But thank you anyway for your input.

closed 2022-12-13 by JuandreG