dchristl/esp32_nat_router_extended #52: esp32c3 version?

pcara, 2023-02-16:
Hi there!

Thanks for this awesome improvement to @martin-ger's nat_router!!

I was trying to compile for my esp32c3 platform but I get the following error after doing

idf.py set-target esp32c3
idf.py menuconfig
[and here putting flash speed to 40MHz and size to 4MB; also LWIP config Layer 2 to Layer 3, IP forwarding etc.]
idf.py build
goes well until

../components/cmd_system/cmd_system.c:228:5: error: implicit declaration of function 'rtc_gpio_isolate' [-Werror=implicit-function-declaration]
  228 |     rtc_gpio_isolate(GPIO_NUM_12);
      |     ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[831/840] Building C object esp-idf/wifi_...f_wifi_provisioning.dir/src/manager.c.objninja: build stopped: subcommand failed.

Any idea of waht I am doing wrong?

Thanks again!

Philippe

dchristl, 2023-02-17:
Hi Philippe,

thank you for your kind feedback. It is hard to say what is wrong there. I usually don't use the esp-idf tools directly, but VSCode with Platform-IO. I have changed some default folders, so eps idf may not find it. But the setup is exactly described [here](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md). Have you already tried it in that way? If not, please try it. You just have to change the platformio.ini - file according to [this documentation](https://docs.platformio.org/en/latest/boards/espressif32/esp32-c3-devkitm-1.html).

Good luck.
Danny

pcara, 2023-02-17:
Hi Danny,

Thanks for the quick reply! I installed the whole VSCode and Platform-IO but I still get the same error:

`...
/home/pcara/ESP/esp32_nat_router_extended/components/cmd_system/cmd_system.c: In function 'deep_sleep':
/home/pcara/ESP/esp32_nat_router_extended/components/cmd_system/cmd_system.c:228:5: error: implicit declaration of function 'rtc_gpio_isolate' [-Werror=implicit-function-declaration]
  228 |     rtc_gpio_isolate(GPIO_NUM_12);
      |     ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[804/813] Building C object esp-idf/cmd_nvs/CMakeFiles/__idf_cmd_nvs.dir/cmd_nvs.c.obj
ninja: build stopped: subcommand failed.`

I still don't see what is wrong :-(

FYI: I did use esp-idf-v5.0 and v4.4, both with same error.

Thanks in advance for any thoughts or ideas.

Philippe

dchristl, 2023-02-18:
Hi Philippe,

I figure it out what the problem is. There is an Option "SOC_RTCIO_HOLD_SUPPORTED", that is not supported on ESP32 C3, so the method rtc_gpio_isolate is not available. But there are more problems. For resettting the device GPIO 23 is used. The C3 doesn't have so many GPIOs, so this will fail, too. 

I've changed these features (dependent on the chip) at the Branch [esp32_c3](https://github.com/dchristl/esp32_nat_router_extended/tree/esp32_C3). You can check if this will work for your device.  I've attached the compiled binary, if you are not able to compile it yourself:
[esp32_c3.zip](https://github.com/dchristl/esp32_nat_router_extended/files/10773961/esp32_c3.zip)

I don't have a C3, so if it works for you, it is possible to merge this and add an C3-release. 

Danny

pcara, 2023-02-18:
Hi Danny,

When I try to compile in the new branch myself it goes much better now but I still get:

/home/pcara/esp/esp32_nat_router_extended/components/cmd_router/cmd_router.c:27:2: error: #error "IP_NAPT must be defined"
   27 | #error "IP_NAPT must be defined"

However... the compiled binaries can be flashed to my esp32c3 without a problem. The device boots and I get the configuration page at 192.168.1.4  Now there, anything I do returns a "Header fields too long" when I submit. So still stuck but already a lot further...

I remember getting the same errors with martin-ger's original esp32_nat_router.

Thanks again for your quick reaction!

Best regards,

Philippe

dchristl, 2023-02-19:
> /home/pcara/esp/esp32_nat_router_extended/components/cmd_router/cmd_router.c:27:2: error: #error "IP_NAPT must be defined"
> 27 | #error "IP_NAPT must be defined"

It seems you are using the wrong sdkconfig. It seems the option CONFIG_LWIP_IPV4_NAPT isn't enabled. 
Did you select the correct target?

[image]

> Now there, anything I do returns a "Header fields too long" when I submit. So still stuck but already a lot further...

I know this problem, I forget to increase the reuest header length in the new sdkconfig. I've changed and pushed this to 
esp32_c3-Branch. See the attached new firmware with changed header size. 
[esp32_c3.zip](https://github.com/dchristl/esp32_nat_router_extended/files/10777067/esp32_c3.zip)

Regards,
Danny

pcara, 2023-02-19:
Hi Danny,

It works now! I can compile myself. Was not using platform.io correctly in the (for me) new environment VScode. Your binaries also work fine.

Now the "Long header file" problem is gone. However now, when I scan the network, I often get an empty list of SSIDs. Strangely enough, always empty on my smartphone and only sometimes on my tablet. So this has probably not much to do with esp32_nat_router_extended... Will experiment more tomorrow.

Thank you so much for your quick and accurate responses! You've earned more than a cup of coffee!

Best regards,

Philippe

pcara, 2023-03-12:
Hi Danny,

I have been testing the nat_router_extended for a few weeks now on my esp32c3. The main functionality (i.e. extending the network) works well and seems to be stable.

The (very handy) wifi scan function now also stopped working on my tablet. This is probably because my chrome webbrowser there got an update. I tested with an older tablet and the old samsung browser on my tablet and there the scan results show a correct list. The newer browsers always show an empty list.

Another thing that does not seem to work for me is de blinking LED. The LED is always on and does not blink to show connection of number of connected hosts. This might be because my ESP32c3 board has a strange onboard LED (R,G,B on gpio 3,4,5, active on LOW) for which I had to change a few things in the code but I also tried with an external LED on gpio 2 with the original code, with same result. I don't see where ap_connect changes value and I can't fix this (with my very limited knowledge of cpp and esp32).

Anyway, as I said, the main functionality works well and I have been using it for two weeks without problems.

Thanks again!

Philippe

dchristl, 2023-03-14:
Hi Philippe,

thank you for your feedback. 

> The (very handy) wifi scan function now also stopped working on my tablet. This is probably because my chrome webbrowser there got an update. I tested with an older tablet and the old samsung [...]

I'm not quite sure what can cause this issue, I have an actual phone and have tested it with multiple Browsers (Chrome, Fennec, Brave) and it works in all of them. Can you please test it with another Browser than Chrome on your phone (i.e. Firefox)? Keep in mind that you can call the result page only one time. After the microcontroller has send the data to a device all scan results will be lost. You have to scan again every time.

> Another thing that does not seem to work for me is de blinking LED. The LED is always on and does not blink to show connection of number of connected hosts. This might be because my ESP32c3 board [...]

This (and the [reset by PIN](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#resetting-the-device-erasing-the-flash)) was an expected issue with other devices. There are a lot of different devices with different configurations out there. The implementation is for the most available ESP32-dev boards out there. Adaption for every device is sadly not maintainable for me. 

Kind Regards
Danny

pcara, 2023-03-15:
Hi Danny,

I did a few more tests with my laptop, where I have better control on the wifi networks connected and there it works. Then I went back to my smartphone and switched off all wifi networks known to my phone. Although I still don't get consistent results I managed to get some proper lists after wifi scan, both in chrome and in firefox. Sometimes after reconnecting my phone goes to the captive portal automatically and when I get back to the proper browser to see the scan results I get an empty list. So maybe there is something wrong with my setup or phone.

As for the LED: don't worry about that. The LED switches on and I can switch it off via the web UI. It is only the blinking that does not work on my board. It is not a big deal.

Best regards,

Philippe

closed 2023-03-26 by pcara

dchristl, 2023-04-30:
@pcara I've release an "official" version for the C3. It will be nice if you test it and give me feedback. Thank you

pcara, 2023-05-14:
hi Danny,
I was a bit busy the past few weeks. Today I tried the C3-image that you provide. I flashes alright but when booting the ESP, no webpage comes up at 192.168.4.1 and when I monitor the serial port I get 

invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header:�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff

repeatedly. So there must be something wrong with the image. 

I have not had time to look at the source though.

Thanks again for all your work!

Regards,

Philippe

dchristl, 2023-05-14:
Hi @pcara,

is it possible, that you have flashed the update-version instead the full one? It seems there is no bootloader installed on your device. The difference between the update and the full version is that in the update version the partitions.bin and the bootloader.bin are not included. Otherwise try to erase the flash before.

Regards,
Danny

PS: I've already ordered a C3 device to test it out for myself. Unfortunately, the device has not arrived yet.

pcara, 2023-05-20:
Hi Danny,
I did flash the full version. I also tries erase_flash.
Today I saw version 5.0.0 and tried that. With the same result :-(

Thanks for all your efforts!
Philippe

pcara@nuc:~/esp/esp32_nat_router_extended$ unzip esp32c3nat_extended_full_v5.0.0.zip 
Archive:  esp32c3nat_extended_full_v5.0.0.zip
  inflating: esp32c3nat_extended_full_v5.0.0.bin  
pcara@nuc:~/esp/esp32_nat_router_extended$ esptool.py erase_flash
esptool.py v4.4
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 10:91:a8:17:bd:2c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 17.4s
Hard resetting via RTS pin...
pcara@nuc:~/esp/esp32_nat_router_extended$ esptool.py write_flash 0x0 esp32c3nat_extended_full_v5.0.0.bin 
esptool.py v4.4
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 10:91:a8:17:bd:2c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00153fff...
Compressed 1389888 bytes to 689517...
Wrote 1389888 bytes (689517 compressed) at 0x00000000 in 61.4 seconds (effective 181.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

pcara, 2023-05-20:
Me again @dchristl !
Trying the C3 branch I manage to build, upload and get things to work... However the result page still does not get displayed.
Monitoring the device, I see the following

I (26752) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (26752) wifi:station: 5e:5b:06:09:4e:2e join, AID=1, bgn, 40U
I (26762) ESP32 NAT router: Station connected
W (27092) wifi:<ba-add>idx:2 (ifx:1, 5e:5b:06:09:4e:2e), tid:0, ssn:0, winSize:64
I (27092) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
W (27122) wifi:<ba-add>idx:3 (ifx:1, 5e:5b:06:09:4e:2e), tid:7, ssn:1, winSize:64
[0;32mI (27232) DNSServer: Received 32 bytes from 192.168.4.2 | DNS reply with len: 48
;32mI (27232) DNSServer: Waiting for data
I (27232) DNSServer: Received 47 bytes from 192.168.4.2 | DNS reply with len: 63
I (27242) DNSServer: Waiting for data
[0;33mW (27312) httpd_uri: httpd_uri: URI '/generate_204' not found
[0;32mI (27322) DNSServer: Received 34 bytes from 192.168.4.2 | DNS reply with len: 50
;32mI (27322) DNSServer: Waiting for data
[0;32mI (27352) cmd_router: scan_result WiFi-2.4-5918-43ROMES-75WiFi-2.4-17F0-76WiFi-2.4-2354-76WiFi-2.4-LE21-79WiFi-2.4-B978-84telenet via devolo-88DIRECT-44-HP ENVY 5000 series-88telenet-04A24-90Orange-45cc2-90SUN2000-HV2280472727-92
I (27372) IndexHandler: Scan result is available. Forwarding to scan page
0;32mI (27382) cmd_router: scan_result WiFi-2.4-5918-43ROMES-75WiFi-2.4-17F0-76WiFi-2.4-2354-76WiFi-2.4-LE21-79WiFi-2.4-B978-84telenet via devolo-88DIRECT-44-HP ENVY 5000 series-88telenet-04A24-90Orange-45cc2-90SUN2000-HV2280472727-92[0m
I (27382) DNSServer: Received 39 bytes from 192.168.4.2 | DNS reply with len: 55m
I (27412) DNSServer: Waiting for data
I (27422) ResultHandler: Requesting result page
[0;32mI (27802) DNSServer: Received 45 bytes from 192.168.4.2 | DNS reply with len: 61
;32mI (27812) DNSServer: Waiting for data
[0;32mI (27842) IndexHandler: Allocating additional 5522 bytes for config page.
[0;32mI (27882) LockHandler: Requesting style.css
[0;33mW (28052) httpd_uri: httpd_uri: URI '/generate_204' not found
[0;32mI (28062) LockHandler: Requesting jquery
[0;32mI (28282) LockHandler: Requesting bootstrap
[0;32mI (28602) LockHandler: Requesting favicon
[0;33mW (28652) httpd_uri: httpd_uri: URI '/generate_204' not found
[0;32mI (31862) DNSServer: Received 39 bytes from 192.168.4.2 | DNS reply with len: 55
;32mI (31862) DNSServer: Waiting for data

Maybe this helps?

Best regards,

Philippe

dchristl, 2023-05-22:
It seems the scan works (Scan result is available. Forwarding to scan page) and you will be forwarded to the page to show the results  (Requesting result page). This also means your browser consumes the page, because this is the last entry before the page will be send back to the client. So, your ESP 32 works like expected and your browser seems not to show the page. Have you any explanation for that?  Do you have the option to use an other browser? Or how do you browse after the reconnection?
The normal workflow is:

- Connecting to the device
- Go to scan page and scan
- Your browser stays open
- Your device gets disconnected from the ESP32
- The ESP32 reboots
- Your device reconnects automatically and your waiting browser will be forwarded to the result
- Or (if your browser was closed) you go to the start page of the ESP32 and you will be autmatically redirected to the result
- After delivery of the result, the ESP32 deletes the result.

If you want to see the result again, you have to rescan.

Best regards,
Danny