dchristl/esp32_nat_router_extended #53: portmap

Orion-55, 2023-03-11:
Hallo erst mal vielen dank für die gelungene Neugestaltung des Nat Routers. Ich habe ausgiebig getestet und es funktioniert einfach toll. Es gibt dennoch ein einziges Problem. Was mir im moment den Einsatz schwer macht. Wenn ich mit den Befehl 
portmap add TCP 8080 192.168.4.2 80 eine Portweiterleitung einrichte, bekomme ich als Antwort über show lediglich folgende Zeile.
TCP 0.0.0.0:8080 -> 192.168.4.2:80

Ich habe daraufhin mit der Version von https://github.com/martin-ger/esp32_nat_router
getestet und erhalte diese Zeile.

TCP 192.168.179.51:8080 -> 192.168.4.2:80 
Diese Weiterleitung funktioniert dann einwandfrei.
Was mache ich bei ihrer Version falsch ?

dchristl, 2023-03-12:
Hallo Orion-55,

danke für deine Rückmeldung und den Bugreport. Tatsächlich wird der portforward und allg. der CLI-Teil von mir etwas vernachlässigt, da ich den selbst nicht nutze. Wahrscheinlich funktioniert  dieser seit einem ESP-IDF-Update nicht mehr. Die IDF-Version, die die Grundlage von Martin-Gers Projekt ist, ist schon relativ alt und ich versuche das Projekt aktuell zu halten und an die neuesten IDF-Versionen anzupassen, ws mal mehr mal weniger Aufwand und potentielle Fehlerquellen bedeutet.

Ich werde mir das Problem anschauen und reparieren. 

Viele Grüße
Danny

Orion-55, 2023-03-12:
Danke für ihre schelle Rückmeldung.
Die version von Martin-Gers ist bis zum gelingen  ein Kompromiss und ich freue mich jetzt schon auf die Neue von ihnen.
Ein AP ist bereits im Dauereinsatz aber da benötige ich auch kein portforward. 

Viele Grüße
Klaus

dchristl, 2023-04-30:
I've release a new version and hopefully fixed this bug. It will be nice if you test it and give me feedback. Thank you

closed 2023-04-30 by dchristl

Orion-55, 2023-05-01:
Hallo, ich habe ihre Version ( esp32nat_extended_full_v4.1.0.bin )
natürlich gleich getestet.

show
[0;32mI (14170) cmd_router: ssid WLAN-72700 [0m
[0;32mI (14170) cmd_router: passwd xxx [0m
[0;32mI (14170) cmd_router: ap_ssid ESP32_Nat_Router [0m
[0;32mI (14180) cmd_router: ap_passwd xxx [0m
STA SSID: WLAN-727 Password: xxx
AP SSID: ESP32_Nat_Router Password: xxx
AP IP address: 192.168.4.1
Uplink AP connected
IP: 192.168.179.139
0 Stations connected
TCP 192.168.179.139:8090 -> 192.168.4.2:80
TCP 192.168.179.139:8091 -> 192.168.4.3:80
TCP 192.168.179.139:8092 -> 192.168.4.4:80
TCP 192.168.179.139:8093 -> 192.168.4.5:80
TCP 192.168.179.139:8094 -> 192.168.4.6:80
TCP 192.168.179.139:8095 -> 192.168.4.7:80
TCP 192.168.179.139:8096 -> 192.168.4.8:80
TCP 192.168.179.139:8097 -> 192.168.4.9:80
TCP 192.168.179.139:8098 -> 192.168.4.10:80
esp32>

Ich bekomme aber leider weiterhin keine Verbindung. Die Webseiten sind auch
jetzt noch nicht erreichbar.
Vielen Dank aber erst ein mal für ihre Bemühungen.
MfG
Klaus

dchristl, 2023-05-02:
Hallo, so wie es aussieht habe ich die falsche Version hochgeladen.🤦 Also eine, wo das Problem noch nicht komplett gelöst ist. Werde die richtige schnellstmöglich nachreichen.

dchristl, 2023-05-02:
Bei der aktuellsten Version 4.1.1 sollte jetzt alles funktionieren.

Orion-55, 2023-05-08:
Hallo, ich habe ihre Version 4.1.1 nun auf drei ESP32 im Einsatz.
Es funktioniert alles ohne Probleme.
Der Password Schutz wird nur nach Reboot aktiv. Ist das so gewollt ?

Vielen Dank.
MfG
Klaus

commit 131bdfd7 (2023-04-30): Fixed issue with portmap closes #53
commit 05ace7d6 (2023-05-02): Missing commit that closes #53