dchristl/esp32_nat_router_extended #54: can provide the boot loader ,firmware,partions bin files separately

RohanSiva28, 2023-03-12:
can provide the boot loader ,firmware, partions bin files separately i tried flashing the bin file u gave but it is of no use when opened serial monitor it is failing to load any bin files and ween tryed to build the project with vscode it is saying it is unable to compile because of white spaces in project path

dchristl, 2023-03-14:
Have you read already [the build section](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md). There is everything explained how to build this project with vscode.
Have you removed the whitespaces in your project path?

RohanSiva28, 2023-03-15:
yes it worked but it is unstable any solution for it works fine for few min but stopes work and i have hit the reset switch to make it work again

dchristl, 2023-03-18:
It seems you are having problems with your power or your network in general (i.e. too long distance, interferences, blocked channel, ... ). Me and some other users use this over a long period of time. Have you aready tried another device or another location? I don't think it is a firmware problem.

RohanSiva28, 2023-03-20:
YES after a few tests.I found out the network(Collage Wi-Fi) i am using is not happy with sharing on 2.4GHz band but its okay on 5GHz . but my esp32 is not capable of 5GHz . can suggest me any model of esp32 which supports 5GHz cause i saw that some versions of esp32 have Wi-Fi 6 .But i am not sure whether this firmware will be supported by those versions it would be great if u have any suggestion .thank you for the responses.

dchristl, 2023-03-27:
Sorry, I can't really suggest one, because I only own some default ones. The [ESP32-C5](https://www.espressif.com/en/news/ESP32-C5) seems to be capable of 5GHz. I'm not quite sure, if this firmware will work, because I am not able to try it. If you have success, it will be great if you to give feedback.

closed 2023-04-30 by dchristl