dchristl/esp32_nat_router_extended #55: ESP32-S2

jonzobot, 2023-03-26:
Trying to flash to an ESP32-S2-MINI-2U. I was able to flash martin_ger's by building and flashing. 

Flashing the binary succeeds, but the wifi doesn't appear, so there's nothing to connect to. I also tried building using idf.py but I'm getting some compilation failures (missing some libraries). Not sure if anyone has flashed to an S2 yet or what idf version works for building.

jonzobot, 2023-04-23:
Thanks, dchristl. Took a peek at the serial output and I guess there's an issue with the partition table. Flashing the v. 4 binary, it leads to a repetition of this output:

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4000f79a
Invalid chip id. Expected 2 read 0. Bootloader for wrong chip?
ets_main.c 386 

I took a quick try to build with idf.py with no joy (different errors based on the version of IDF but couldn't find a working one); for whatever reason I can't get VSCode to work either but I have never used that software before. I'm a bit low on free time these days so at some point I'll try it again.

dchristl, 2023-04-30:
I think this is the same problem like #52 . There is a need for a separate configuration file for the S2, because I use some features (i.e. reset, LED) that are not available on all chips. You can try the C3-version, uploaded in the linked thread or I will test and provide a special S2-version.

jonzobot, 2023-05-04:
Thanks, dchristl. I just flashed the C3 version 4.1.1, but no dice. I wind up getting the following error after a successful flash:

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Invalid chip id. Expected 2 read 5. Bootloader for wrong chip?
ets_main.c 386 

I'll see if I can build for an S2 board in the coming days.

dchristl, 2023-05-05:
Ok, the error is because of the wrong chip version, that is set into the single binary. Please try the [attached version](https://github.com/dchristl/esp32_nat_router_extended/files/11407154/single_file.zip), that has some changes for the S2. If it doesn't work, please try to flash bootloader, firmware and partitions separately with 
[this file](https://github.com/dchristl/esp32_nat_router_extended/files/11407153/multiple_files.zip)
The addresses are:
0x1000 -> bootloader
0x8000 -> partitions
0x10000 -> firmware

If you give me feedback, I can build an "official" version as a release

jonzobot, 2023-05-06:
Hey, it worked! Great! 

And it's faster too, I'm getting about 5 mbps as opposed to ~2 that I was before.

dchristl, 2023-05-06:
Great, I will add it to the releases. 
Just one question: Have you used the single binary or the splitted binaries?

jonzobot, 2023-05-08:
I used the single binary. Cheers.

closed 2023-05-28 by dchristl