dchristl/esp32_nat_router_extended #59: Doccument request!

wangyiwu, 2023-05-03:
First of all I would like to thank you for the great work you did! Can you guide the configuration to build the project on Eclipse instead of on PlatformIO, thank a lot!

wangyiwu, 2023-05-04:
I already can build on Eclipse by https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md

dchristl, 2023-05-05:
Hello @wangyiwu,

sorry, but I can't help you with that. I have little to no experience with Eclipse and I think PlatformIO is better for a development with different boards.

Your compilation problem is a result of different folder structure (you are not compiling the source folder, so the application finds no entry point). 
If you figure it out, how to compile with Eclipse, you can send me pull request and I can integrate your documentation.

wangyiwu, 2023-05-05:
> Hello @wangyiwu,
>
> sorry, but I can't help you with that. I have little to no experience with Eclipse and I think PlatformIO is better for a development with different boards.
>
> Your compilation [...]

I think I maybe related to this issue: https://github.com/espressif/esp-idf/issues/8974

I will create a document for compile with Eclipse,  For people who are interested in working on Eclipse IDE

wangyiwu, 2023-05-06:
do you have nay suggestion for those error: 
[image]

dchristl, 2023-05-06:
> I think I maybe related to this issue: https://github.com/espressif/esp-idf/issues/8974

I don't think this is an issue with esp-idf. This is a common problem with wrong setups. 

> do you have nay suggestion for those error:
Your 2nd error is because of the wrong Esp-Configuration (sdkconfig). 

Why do you want to compile with Eclipse so badly? I think you have to modify too much just to get it to work. With PlatformIO, you have a solution that works out of the box.

wangyiwu, 2023-05-06:
> > I think I maybe related to this issue: [espressif/esp-idf#8974](https://github.com/espressif/esp-idf/issues/8974)
> 
> I don't think this is an issue with esp-idf. This is a common problem with wrong setups.
> 
> > do you have nay suggestion for those error:
> > Your 2nd error is because of the wrong Esp-Configuration (sdkconfig).
> 
> Why do you want to compile with Eclipse so badly? I think you have to modify too much just to get it to work. With PlatformIO, you have a solution that works out of the box.

yes, My personal opinion is that Eclipse is not a very good IDE, but during the setup process it helps me understand how the compile process will happen. That's why I still choose Eclipse.

dchristl, 2023-05-06:
OK, I see, but I don't think there is really a difference. Both uses the python-tools by espressif (esp-idf, esptool, ...) for compilation, deployment, monitoring and the output is the same. The only difference is the editor.

closed 2023-12-15 by dchristl