dchristl/esp32_nat_router_extended #62: How to select a wifi channel?

PE2016, 2023-05-12:
Great work, thanks for providing binaries. I followed your instructions on an ESP32 DivKit, everything works. However not sure how it would work as wifi e cf tender!

Now I got the new ESP32 C5 that supports 5 GHz; can I use the same binaries?
Is it possible to add the capability of changing the frequency channel?
Thank you in advance

dchristl, 2023-05-12:
Hello @PE2016 ,

thank you for your feedback. 

> Now I got the new ESP32 C5 that supports 5 GHz; can I use the same binaries?

This will not work, because there are some differences, but I can try to provide you with an image to try out.

> Is it possible to add the capability of changing the frequency channel? 

No yet, currently it checks for the best one nad uses this dynamically.

PE2016, 2023-07-04:
Greetings,
I am reaching to you regarding your NAT version 5.1, I had flashed your
binary and it was working fine, but then it stopped!

I erased and flashed my ESP 32 with the esp32nat_extended_full_v5.1.0
binary. I used the latest flash_download_tool_3.9.5 as shown below, but I
never got it to work!
Then, I got another similar chip, and flashed it with version 4.0 and the
new chip didn't work either!
What am I doing wrong? Any advice is appreciated,
Thanks
Ramez

Version 5.1

[image: image.png]

Version 4
[image: image.png]

dchristl, 2023-07-07:
Hello,

could you provide the images? It doesn't seem to work, because they are not visible. Normal flashing on Windows is done with:
https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/win_flash_full.png

What kind of chip do you have? Is it a C5, than I don't think this will work. Otherwise is it possible to provide a logfile?

Regards,
Danny

PE2016, 2023-07-07:
Hi DannyNo, both chips are Esp32-S. BestRamezSent from my iPhone

https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/win_flash_full.png

PE2016, 2023-07-07:
Hi Danny
No, both chips are Esp32-S. 
Best
Ramez

Sent from my iPhone

dchristl, 2023-07-08:
Sorry, but I'm unable to help you if you don't provide the requested informations. Your pictures are not visible, you didn't provide a log and there two ESP32 S (S2 and S3). Which one is yours

PE2016, 2023-07-08:
Danny

Sorry that my email was not clear.

I am using the  ESP-WROOM-32 module. It is an ESP32-S, and  not the S2 or
S3.

Attached are the log files for v4 & v5 of your code.

Thank you,

Ramez

*************************** START ****************************
START TIME: 202307081611
CONNECT BAUD: 115200
set state: ESP_DL_SYNC
serial port opened
-----------
baud:115200
root baud:115200
-------------
===============BAUD : 115200===============CALL DEVICE SYNC
connecting...
mac l: 0x5325528
mac h: 0x353c61
crc_cal: 53
crc_read: 53
crc_test: 0
ESP32 MAC CRC OK
get mac res: 1
get flash id : 0x001640d8
manufacturer_id: 0xd8
device_id: 0x4016
Wrote 1320096 bytes (711360 compressed) at 0x00001000 in 63.4 seconds (effective 166.5 kbit/s)...
Hash of data verified.
Leaving...
com closed
Use time: 69 s
Download result: PASS
*************************** END ****************************


*************************** START ****************************
START TIME: 202307081615
CONNECT BAUD: 115200
set state: ESP_DL_SYNC
serial port opened
-----------
baud:115200
root baud:115200
-------------
===============BAUD : 115200===============CALL DEVICE SYNC
connecting...
mac l: 0x1fc66ff4
mac h: 0x2d9c9c
crc_cal: 45
crc_read: 45
crc_test: 0
ESP32 MAC CRC OK
get mac res: 1
get flash id : 0x001640d8
manufacturer_id: 0xd8
device_id: 0x4016
Wrote 1346880 bytes (733295 compressed) at 0x00001000 in 65.4 seconds (effective 164.7 kbit/s)...
Hash of data verified.
Leaving...
com closed
Use time: 138 s
Download result: PASS
*************************** END ****************************

dchristl, 2023-07-09:
> I am using the  ESP-WROOM-32 module. It is an ESP32-S, and  not the S2 or
S3.

There is no S-only device. Can you provide a link or an image of your board? But I think it is a S2, otherwise the flashing will result in an error. Your flashing process seems fine overall. The only things I can imagine is that you have used not the full version (esp32s2nat_extended_full_v5.1.1.zip).  
I've checked with this version and everythings works like expected with these settings:
[image]

Have you rebooted your device afterwards, respectively leaved the download mode? This is different between the devices, but a reboot will do it.

PE2016, 2023-07-09:
DannyThank you for your prompt response.Yes, I tried reboot and reset too.Attached is the ESP32 images,BestRamezSent from my iPhone

dchristl, 2023-07-09:
Could you please check this thread on Github and not with your IPhone. Please answer there and not by mail. None of your images are visible?

PE2016, 2023-07-09:
[image]
[image]

PE2016, 2023-07-09:
I flashed it again with version 5.1.1; same outcome! Here he is the ESP32  #image.
[image]
[image]
[9C9C1FC66FF4.txt](https://github.com/dchristl/esp32_nat_router_extended/files/11995920/9C9C1FC66FF4.txt)

dchristl, 2023-07-10:
I really don't know what kind of chip this is, but is not an regular S2 or S3, because they don't have Bluetooth. It seems to be a normal ESP32. You have two patch-options now. You can try to flash the normal esp32nat_extended_full_v5.1.1.zip-package and have a look if this will work. The other option is to patch the [predecessor project](https://github.com/martin-ger/esp32_nat_router) and try this out. Please provide logs for the patching process. The next what you can try is to establish a serial connection [with Putty](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html) after the flash and show me the output. 

Otherwise I'm afraid this chip is not supported. Do you have a datasheet for it?

PE2016, 2023-07-12:
It is an ESP32 as reported when I used the esptool.py to flash v5.1.1!
It still didn't work, it reported an error when I tried Putty, as per the attached. 

[NAT-Putty-Esptool output.pdf](https://github.com/dchristl/esp32_nat_router_extended/files/12022745/NAT-Putty-Esptool.output.pdf)

dchristl, 2023-07-12:
It seems it is just a relabelled ESP32(without the S). And I see the problem here. You didn't flash as described. You try to install a full version on the wrong address. The address 0x1000 is just for update versions, to avoid a new setup. You have to flash it at address 0x0 instead. Have a look at the documentation (section flashing the prebuild binaries ) for further informations.

closed 2023-07-12 by PE2016

dchristl, 2023-07-13:
Your problem has nothing to do with this project, but with your power supply. There are a lot errors in both of your logs (brown out detected ) that causing a restart. Change your cable or whatever power source you have and both controller will work fine.