dchristl/esp32_nat_router_extended #64: WPA3 ?

ErfanDL, 2023-05-27:
Hi @dchristl 
is it possible to add WPA3 support ?
thanks.

dchristl, 2023-05-28:
Hello @ErfanDL,

you are talking about the AP settings, right, because connecting to WPA3 routers should already work out of the box. Last time I checked this was not supported in softAPs, but it seems to be [partially supported](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/wifi-security.html) now.
But I'm not quite sure about to add it, because there are some downsides. Some devices doesn't support this, what causes  lesser compatibility. if you have a good WPA2 password the security is nearly the same like WPA3. I will check out if the mixed mode (WPA2/WPA3) will work well. 

Do you have any specialities, to have this supported? 

Regards,
Danny

ErfanDL, 2023-05-28:
Thank for reply, of course it doesn't matter too much, since I have a Mikrotik router and it is set to WPA3, I was curious to know if it is possible to use WPA3 on the ESP32 board or not.

dchristl, 2023-05-28:
To clear this up: If your Router is WPA3 the connection between this and the ESP32 is already WPA3. Only the connection between an other device, connected to the ESP, is WPA2 only. I've researched it and a WPA3 mode with WPA2 compatibility is supported. So, I will test it out and implement it, if it is stable. There were some flaws in the past with WPA3.

dchristl, 2023-05-29:
I've tried it out and the current state is like before. WPA3 in softAP-mode (which is needed to make all this work) is not supported in the latest stable esp-idf (currently 5.0.2). But, in the upcoming release (5.1) this should work according to the documentation. I will add it, when it works stable.

dchristl, 2023-08-12:
This feature is now available since [v6.3.0](https://github.com/dchristl/esp32_nat_router_extended/releases/latest)

closed 2023-08-12 by dchristl