dchristl/esp32_nat_router_extended #67: Improvement: scheduled reboot

ulipo, 2023-06-05:
I've experienced that sometimes my esp32 router becomes sluggish or irresponsive and needs a reboot (probably due to the high number of clients or a bad PSU, dunno...).
Since a reboot just takes a couple of seconds, it is not a big deal for the clients, that promptly reconnects; I would love to be able to schedule a daily reboot (or maybe after N hours from boot) without having to reach the router by hand.

dchristl, 2023-06-07:
Hello @ulipo ,

interesting idea, the problem is that implementing this small functionality would require a lot of libraries and code. While the ESP32 has a Real-Time Clock, it unfortunately lacks cron functionality. The time would need to be synchronized using NTP, and additional settings for time zone and cron syntax would have to be added. All this just for triggering a restart. Could you please provide further analysis of your problem? Perhaps there is an alternative way to trigger a restart, such as high CPU load. That would be easier to implement. 

If one of your clients is a Linux client and it's constantly running, we could already achieve something with cron and curl.

Regards,
Danny

ulipo, 2023-06-07:
I was thinking more to something really simple like "when uptime is bigger than X, reboot" 
I have a NUC with Debian always running on the same network, I guess I can just call "http://DEVICE_IP/apply" with a cron job and keep my ESP32 freshly booted and responsive :-)

Regarding how and why it becomes sluggish, I have no clue... If you have any idea on how to log some data, I am available to try.

dchristl, 2023-06-08:
> I was thinking more to something really simple like "when uptime is bigger than X, reboot"

This one will be easy, I will think about it. The disadvantage is that the restart always depends on the initial start. So if you start the system at 12 noon, the restart will also be at that time.

> I guess I can just call "http://DEVICE_IP/apply" with a cron job and keep my ESP32 freshly booted and responsive :-)

That is exactly what was in my mind ;)

> Regarding how and why it becomes sluggish, I have no clue... If you have any idea on how to log some data, I am available to try.

Maybe you were connected to the terminal at that time and could see something unusual in the log, or the chip is getting very hot.

fleinze, 2024-06-03:
I would need this too, as I have a public network with a timeout of about 60 minutes. With random mac and reboot it works again, but I want to automate this.
how could this be implemented? I would think of adding an option to the advanced page and then calling restartByTimerinS() at startup accordingly. Would this work in your opinion?

dchristl, 2024-06-10:
Hello,

the option in the advanced page is a good start. But normally you would use a timer (performance reason) and not a running thread for 60 minutes. The implementation could look something like this: (untested!!):

```

void timer_callback(void* arg)
{
   //restart ESP32
}

    const esp_timer_create_args_t timer_args = {
        .callback = &timer_callback,
        .name = "restartTimer"
    };

    esp_timer_handle_t timer;
    ESP_ERROR_CHECK(esp_timer_create(&timer_args, &timer));

    // Start the timer (60 minutes = 60 * 60 * 1000000 microseconds)
    ESP_ERROR_CHECK(esp_timer_start_once(timer, 60 * 60 * 1000000));

```

It is not really much to do, but I sadly have no time for that currently.