dchristl/esp32_nat_router_extended #69: Errors :  ESP32 NAT router: disconnected - retry to connect to the AP

pcouderc12, 2023-06-24:
Using the toolchain described in #68, I get frequent disconnections, see full listing below.
I suspect a parameter is missing in configuration, and that  :
*Component config -> LWIP > [x] Enable copy between Layer2 and Layer3 packets.
*Component config -> LWIP > [x] Enable IP forwarding.
*Component config -> LWIP > [x] Enable NAT (new/experimental).

are not enough.

Full log for a connect from an android device: 

I (33) boot: ESP-IDF v5.2-dev-703-gefe919275e 2nd stage bootloader
I (33) boot: compile time Jun 24 2023 10:04:24
I (35) boot: Multicore bootloader
I (39) boot: chip revision: v3.0
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00190000
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=6bdb8h (441784) map
I (267) esp_image: segment 1: paddr=0007bde0 vaddr=3ff80063 size=00008h (     8) load
I (267) esp_image: segment 2: paddr=0007bdf0 vaddr=3ffb0000 size=03980h ( 14720) load
I (279) esp_image: segment 3: paddr=0007f778 vaddr=40080000 size=008a0h (  2208) load
I (282) esp_image: segment 4: paddr=00080020 vaddr=400d0020 size=b8c34h (756788) map
I (563) esp_image: segment 5: paddr=00138c5c vaddr=400808a0 size=168c8h ( 92360) load
I (602) esp_image: segment 6: paddr=0014f52c vaddr=400c0000 size=00064h (   100) load
I (614) boot: Loaded app from partition at offset 0x10000
I (614) boot: Disabling RNG early entropy source...
I (625) cpu_start: Multicore app
I (626) cpu_start: Pro cpu up.
I (626) cpu_start: Starting app cpu, entry point is 0x400813fc
0x400813fc: call_start_cpu1 at /opt/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (643) cpu_start: Pro cpu start user code
I (643) cpu_start: cpu freq: 160000000 Hz
I (644) cpu_start: Application information:
I (648) cpu_start: Project name:     esp32_nat_router
I (654) cpu_start: App version:      v5.1.0-dirty
I (659) cpu_start: Compile time:     Jun 24 2023 10:02:08
I (665) cpu_start: ELF file SHA256:  aac433574...
I (671) cpu_start: ESP-IDF:          v5.2-dev-703-gefe919275e
I (677) cpu_start: Min chip rev:     v0.0
I (682) cpu_start: Max chip rev:     v3.99
I (687) cpu_start: Chip rev:         v3.0
I (692) heap_init: Initializing. RAM available for dynamic allocation:
I (699) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (705) heap_init: At 3FFB83C0 len 00027C40 (159 KiB): DRAM
I (711) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (717) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (724) heap_init: At 40097168 len 00008E98 (35 KiB): IRAM
I (732) spi_flash: detected chip: generic
I (735) spi_flash: flash io: dio
W (739) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (753) app_start: Starting scheduler on CPU0
I (757) app_start: Starting scheduler on CPU1
I (757) main_task: Started on CPU0
I (767) main_task: Calling app_main()
I (817) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (817) ESP32 NAT router: Device will boot up normally.
I (837) ESP32 NAT router: Command history disabled
I (847) cmd_router: ssid TP-LINK_2072
I (847) cmd_router: passwd
I (857) ESP32 NAT router: Netmask is set to 255.255.255.0, therefore private IP is 192.168.4.1
I (877) wifi:wifi driver task: 3ffc23c8, prio:23, stack:6656, core=0
[... 105 lines cut ...]
I (40047) ESP32 NAT router: retry to connect to the AP
I (42887) ESP32 NAT router: disconnected - retry to connect to the AP
I (42887) ESP32 NAT router: retry to connect to the AP
I (43647) DNSServer: Received 39 bytes from 192.168.4.2 | DNS reply with len: 55
I (43647) DNSServer: Waiting for data
I (45727) ESP32 NAT router: disconnected - retry to connect to the AP
I (45737) ESP32 NAT router: retry to connect to the AP
I (48567) ESP32 NAT router: disconnected - retry to connect to the AP
I (48577) ESP32 NAT router: retry to connect to the AP
I (51417) ESP32 NAT router: disconnected - retry to connect to the AP
I (51417) ESP32 NAT router: retry to connect to the AP
I (54247) ESP32 NAT router: disconnected - retry to connect to the AP
I (54257) ESP32 NAT router: retry to connect to the AP
I (57087) ESP32 NAT router: disconnected - retry to connect to the AP
I (57097) ESP32 NAT router: retry to connect to the AP
I (59927) ESP32 NAT router: disconnected - retry to connect to the AP
I (59937) ESP32 NAT router: retry to connect to the AP
I (62767) ESP32 NAT router: disconnected - retry to connect to the AP
I (62777) ESP32 NAT router: retry to connect to the AP
I (65607) ESP32 NAT router: disconnected - retry to connect to the AP
I (65607) ESP32 NAT router: retry to connect to the AP
I (68447) ESP32 NAT router: disconnected - retry to connect to the AP
I (68447) ESP32 NAT router: retry to connect to the AP
I (71287) ESP32 NAT router: disconnected - retry to connect to the AP
I (71287) ESP32 NAT router: retry to connect to the AP

dchristl, 2023-06-30:
Hello @pcouderc12 ,

it is not so easy to build with the native toolchain, because this project is build with platformio. You have to change the whole layout (folders, includes, setting, ...) to make this work. Have you already seen [my instructions how to build](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md)? I will not recommend and support other builds than with platformio. 

Regards,
Danny

pcouderc12, 2023-06-30:
Sorry that your are wrong. It is very easy to build with native toolchain and I have given the way to do that in #68...
But you are free to not recommand and not support it. And to not try answering my problem.

dchristl, 2023-06-30:
I know what to do, because I've done it the other way round when I started this project. ;) In my opinion it is not very easy, because you have to do these steps everytime when there is any code change in this project. Otherwise you get a lot of merge conflicts. It is not that I don't want to help you, but I can't and don't recommend it. I don't see your setup, because maybe there is something wrong. Do you have this problems when building with platformio or using my release, too?
Btw. what is the advantage by building with the native toolchain, because platformio uses this under the hood, too. But with a uniformed folder structure (same for every supported microcontroller, independent from vendor's toolchain)

pcouderc12, 2023-06-30:
All and the only setup I have done is indicated in #68...
And this is certainly the problem, I am missing something.
I do not use platformio... because I have started with native toolchain and did see no interest in other  environnement. And I am sure I may be wrong...
Thanks anyway !

dchristl, 2023-07-01:
The only thing what you can try is to replace the content of your sdkconfig with mine, dependent on your chip (i.e. sdkconfig.esp32dev for esp32), because I changed more settings than just the ones you mentioned (i.e CONFIG_HTTPD_MAX_REQ_HDR_LEN). 
The second option is to use one of my binaries to exclude issues with your chip or any issues with my code.

pcouderc12, 2023-07-01:
Thank you, these are 2 interesting ways. My chip couldbe a problem.

pcouderc12, 2023-07-01:
There is some mystery in  my installation, as I have changed the esp32 device, and even the devkit. And I have tried with esp32nat_extended_full_v5.1.0.bin and even Martin Ger binary. Every time, I success connecting  to http server to program the router, but after  reset, I get AP disconnections.... No more ideas now.

dchristl, 2023-07-01:
Could you give me an actual log?

pcouderc12, 2023-07-01:
Sure !
log done under debian using minicom, after erasing flash with :
esptool.py --chip esp32 erase_flash
flashing with:
esptool.py write_flash 0x0  esp32nat_extended_full_v5.1.0.bin
then starting minicom
then connecting debian to ESP32_NAT_Router
then filling parametes with http server (open networls), then restarting with hhtp server :

I (580) cpu_start: Pro cpu up.
I (580) cpu_start: Starting app cpu, entry point is 0x40082188
I (0) cpu_start: App cpu up.
I (594) cpu_start: Pro cpu start user code
I (594) cpu_start: cpu freq: 160000000 Hz
I (594) cpu_start: Application information:
I (599) cpu_start: Project name:     esp32_nat_router
I (605) cpu_start: App version:      134c8ce-dirty
I (610) cpu_start: Compile time:     Jun  2 2023 15:39:03
I (616) cpu_start: ELF file SHA256:  8db8daf4413a8365...
I (622) cpu_start: ESP-IDF:          5.0.2
I (627) cpu_start: Min chip rev:     v0.0
I (632) cpu_start: Max chip rev:     v3.99
I (637) cpu_start: Chip rev:         v3.0
I (642) heap_init: Initializing. RAM available for dynamic allocation:
I (649) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (655) heap_init: At 3FFB7D68 len 00028298 (160 KiB): DRAM
I (661) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (667) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (674) heap_init: At 4009760C len 000089F4 (34 KiB): IRAM
I (681) spi_flash: detected chip: generic
I (685) spi_flash: flash io: dio
I (690) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (724) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (734) ESP32 NAT router: Device will boot up normally.
I (754) ESP32 NAT router: Command history disabled
I (754) ESP32 NAT router: Netmask is set to 255.255.255.0, therefore private IP is 192.168.4.1
I (774) wifi:wifi driver task: 3ffc2500, prio:23, stack:3584, core=0
I (774) system_api: Base MAC address is not set
I (774) system_api: read default base MAC address from EFUSE
I (784) wifi:wifi firmware version: 57982fe
I (784) wifi:wifi certification version: v7.0
I (794) wifi:config NVS flash: enabled
I (794) wifi:config nano formating: disabled
I (804) wifi:Init data frame dynamic rx buffer num: 32
I (804) wifi:Init management frame dynamic rx buffer num: 32
I (804) wifi:Init management short buffer num: 32
I (814) wifi:Init dynamic tx buffer num: 32
I (814) wifi:Init static rx buffer size: 1600
I (814) wifi:Init static rx buffer num: 10
I (824) wifi:Init dynamic rx buffer num: 32
I (834) wifi_init: rx ba win: 6
I (834) wifi_init: tcpip mbox: 32
I (834) wifi_init: udp mbox: 6
I (844) wifi_init: tcp mbox: 6
I (844) wifi_init: tcp tx win: 5744
I (844) wifi_init: tcp rx win: 5744
I (854) wifi_init: tcp mss: 1440
I (854) wifi_init: WiFi IRAM OP enabled
I (854) wifi_init: WiFi RX IRAM OP enabled
I (884) ESP32 NAT router: set dns to: 192.168.4.1
I (2884) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (2884) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (3044) wifi:mode : softAP (cc:db:a7:1e:3f:29)
I (3044) wifi:Total power save buffer number: 16
I (3044) wifi:Init max length of beacon: 752/752
I (3054) wifi:Init max length of beacon: 752/752
I (3064) ESP32 NAT router: wifi_init_ap with default finished.
I (3064) DNSServer: DNS Server started
I (3064) DNSServer: Socket created
[... 170 lines cut ...]
I (25748) ESP32 NAT router: disconnected - retry to connect to the AP
I (25758) ESP32 NAT router: retry to connect to the AP
I (28588) ESP32 NAT router: disconnected - retry to connect to the AP
I (28598) ESP32 NAT router: retry to connect to the AP
I (31438) ESP32 NAT router: disconnected - retry to connect to the AP
I (31438) ESP32 NAT router: retry to connect to the AP
I (34278) ESP32 NAT router: disconnected - retry to connect to the AP
I (34278) ESP32 NAT router: retry to connect to the AP
I (37118) ESP32 NAT router: disconnected - retry to connect to the AP
I (37118) ESP32 NAT router: retry to connect to the AP
I (39958) ESP32 NAT router: disconnected - retry to connect to the AP
I (39968) ESP32 NAT router: retry to connect to the AP
I (42798) ESP32 NAT router: disconnected - retry to connect to the AP
I (42808) ESP32 NAT router: retry to connect to the AP
I (45648) ESP32 NAT router: disconnected - retry to connect to the AP
I (45648) ESP32 NAT router: retry to connect to the AP
I (48488) ESP32 NAT router: disconnected - retry to connect to the AP
I (48488) ESP32 NAT router: retry to connect to the AP
I (51328) ESP32 NAT router: disconnected - retry to connect to the AP
I (51328) ESP32 NAT router: retry to connect to the AP
I (54168) ESP32 NAT router: disconnected - retry to connect to the AP
I (54178) ESP32 NAT router: retry to connect to the AP
I (57018) ESP32 NAT router: disconnected - retry to connect to the AP
I (57018) ESP32 NAT router: retry to connect to the AP
.....

dchristl, 2023-07-02:
Sadly, I don't see something suspicious in your log. Do you have the chance to use an other ESP32 or to connect to another router (i.e. mobile hotspot) with and without encryption?

pcouderc12, 2023-07-03:
Thank you again. I had soon tried to change device. But using an Androïd hotspot, your provided esp32nat_extended_full_v5.1.0.bin works perfectly...  Maybe it is the strenght of the STA signal, even if I do not see why problem with the STA implies a problem with with  AP...

dchristl, 2023-07-03:
The funny part is, that this is just an mistake in the log, that I have already carried over from the original project  ;) This isn't the AP, but the STA. So, this 
> ESP32 NAT router: disconnected - retry to connect to the AP
 ESP32 NAT router: retry to connect to the AP

have to be:
> ESP32 NAT router: disconnected - retry to connect to the STA
 ESP32 NAT router: retry to connect to the STA

Therefore, it is quite possible that it is due to the distance or poor connection. When you scan, is the access point included?

pcouderc12, 2023-07-03:
Mmm, well when scanning, I see and I test with distance from Android augmenting more or less,and so stregnth of signal varying. I have good results with strgnth of signl very low (lower than with the  TP-LINK_2872  open router).
Anyway, I do not understand  why when there is this "ESP32 NAT router: disconnected - retry to connect to the STA" message, I cannot access to 192.168.4.1 page...

pcouderc12, 2023-07-03:
In fact, I do not see in esp32_nat_router.c where wifi_config.sta.authmode is set to  WIFI_AUTH_OPEN when passwd is less than 8  chars. I could explain my problem with my  STA which is open !

pcouderc12, 2023-07-05:
After changing the open router from  WIFI_AUTH_OPEN  to default with password, works perfectly.

closed 2023-07-05 by pcouderc12

dchristl, 2023-07-07:
Great to hear that it works perfectly now. 

> In fact, I do not see in esp32_nat_router.c where wifi_config.sta.authmode is set to WIFI_AUTH_OPEN when passwd is less than 8 chars. It could explain my problem with my STA which is open !

This is normally not needed, because the router and the ESP32 will figure this out for themselves. Maybe something has changed in the framework or it doesn't work with all routers.

> After changing the open router from WIFI_AUTH_OPEN to default with password, works perfectly.

What is the current state now, because you have opened another issue (#70 ) Does it work without these code changes  or onyl when you compile it for yourself? If you need the code changes I will have a deeper look into this. 

Thank you

pcouderc12, 2023-07-07:
Thank you, I did open it, before I had the idea to reprogram the TP_LINK router. And as, after reprogramming this router, the problem is not reproducible, so no need to look for a ghost, and I did close it. Thank you very much for your help. I prepare a version of the routing engine without any dialog, roughly only wifi(...)  and no http or console. For me  and for anyone if it helps...