dchristl/esp32_nat_router_extended #72: MAC Custom setting is not completely correctly

muselc, 2023-07-14:
When I set the MAC address to "78:cf:2f:95:e4:10", it actually sets it to "78:cf:2f:95:e4:11".

[image]

dchristl, 2023-07-15:
Hello @muselc ,

thank you for reporting this issue. Unfortunately, I am unable to replicate the issue. I just tried it and the behavior is as expected:

[image]

The address behind "Your current MAC..." is read out directly form the chip and not from the settings. Which chip are you using, and are you using the latest version of the NAT router? 

Regards,
Danny

muselc, 2023-07-18:
hello @dchristl ,
Thank you for your response. I had tried deploying the latest code on both Win11 and macOS systems, but the issue still persists.I'm sorry, I can't confirm the issue with the chip as I have only one.Below is my device information

## Chip

esp32-WROOM-32

## OS

MacBook Pro M2 Ventura 13.4
Windows 11

dchristl, 2023-07-18:
OK, I see now where the problem lies. This is not a firmware issue but rather a misunderstanding. The ESP32 has 2 Wi-Fi network interfaces in the mode that is needed to get this to work: one for the Access Point and one for the Station (e.g., a router). As a result, it also has 2 MAC addresses. The configuration you can adjust is for the connection to the router, which makes the most sense since there might be MAC filtering settings there. It seems that you are currently connected to your ESP32 (the Access Point) with your device, and you are reading the MAC address from there. Is that correct? This address is not configurable separately and is always 1 higher than the base address used for the connection to the router. If you want to dig deeper in this, have a look at the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html)

What is actually the use case for why you want to change this address?

muselc, 2023-07-19:
I just find it strange that the MAC address is incremented by one.I misunderstood that the `Custom base MAC` is the same as the `Ap MAC`.
From the [release-v5.0](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-reference/system/misc_system_api.html), `esp_netif_set_mac` can do it.
From the [release-v5.1](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/system/misc_system_api.html), `esp_iface_mac_addr_set ` can do it.

closed 2023-07-19 by muselc