dchristl/esp32_nat_router_extended #74: Feature request - wireguard

TimDowker, 2023-07-27:
Any chance you could add wireguard to the image?  Would be great to have a super portable VPN solution like that.

Great work!

dchristl, 2023-07-28:
Hello @TimDowker ,

I think the idea is great. There would be quite a bit to do for that, and I don't think I'll have time for it in the short term. However, I'll keep it in mind, and if I get the chance, I'll expand the router. On the other hand, I am open to accepting Pull requests ;)

Kind regards,
Danny

vkhurana, 2023-11-30:
ESPHome added a WireGuard component that works on the ESP32 (https://esphome.io/components/wireguard.html) so unless there are other constraints it seems like this should be possible. 

Cool idea!

dchristl, 2023-12-01:
Hello @vkhurana,

thank you for the tip; I've also played around with Wireguard on the ESP32. In principle, it should be possible, but there's quite a bit to do, and all available libraries are in alpha, supporting only a few features. So, the functionality will be somewhat limited, and when I have time, that's the next point where I plan to work more intensively.

Kind regards,
Danny