dchristl/esp32_nat_router_extended #75: ESP32-C3 command line interface not working

jimmy-ck, 2023-08-07:
Hi,
at first: Thank you very much for your work – it's working since weeks at my place using a ESP32-CAM (ESP32-S) and a "portmap" for accessing a device behind it. I was able to store this portmap exposing client's port 80 to the router's port 8080 using the following command via serial connection: 
`portmap add TCP 8080 192.168.4.2 80`
This command then gets acknowledged by the ESP with a corresponding answer.

Today I wanted to switch to a ESP32-C3 since I had other uses for the CAM module. Unfortunately though I cannot send any commands via serial to the device. It just ignores everything. Do you have an idea why the complete command line interface works perfectly on ESP32-S but not on ESP32-C3?

Completely different approach: Is it possible to include the portmap feature in the webserver's advanced configuration page?

The command line interface is documented here: [https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#command-line-interface](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#command-line-interface)

dchristl, 2023-08-07:
Hello @jimmy-ck ,

I really don't see a difference in the CLI (and the portmap-feature) between the two devices. The source is completely the same at this part of the application. On which OS are you on and which serial client do you use? Do other CLI-commands (i.e set_sta, show) work or only the portmap ones not? Can you try another client?

> Completely different approach: Is it possible to include the portmap feature in the webserver's advanced configuration page?

I'm currently working on this feature ;) This will be available in the next release. 

Kind regards,
Danny

jimmy-ck, 2023-08-07:
Hi Danny,

I used three different clients (e.g. Arduino IDE) on macOS 13.4.1 – all showing the same behaviour: Send commands are completely ignored. But messages from the ESP are received properly. Any CLI command does not work (help, show, ...). The exact same client works perfectly on a 6.2.0 ESP-32-CAM.
Very likely a behaviour of the C3 itself. I'll dive into that a little and report if I find anything. Thanks for checking for any possible code differences.

> I'm currently working on this feature ;) This will be available in the next release.

That's great! Looking forward for this. :)

BR,
Christopher

dchristl, 2023-08-08:
Maybe there is a difference not in the chip itself, but in the way the developmnet board  manages the serial connection. It seems there is a issue with your combination and I found [this Reddit thread](https://www.reddit.com/r/esp32/comments/pf61v1/esp32c3_serial_output_in_arduino_ide/?rdt=58248). Maybe this can help. Otherwise use the built in **screen** command, like described [in the official docs](https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32c3/get-started/establish-serial-connection.html). I've tried it (under Linux) and it works like a charm:

[image]

jimmy-ck, 2023-08-08:
I stumbled onto something similar in some forums – I'm using the XIAO ESP-32-C3 ([seeed studio wiki](https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/)).
I tried now using a Windows PC and PuTTY now – same problem. Interesting that you got it running on linux. I'll try that later that day. Thank you for your research!

jimmy-ck, 2023-08-08:
Seems like there might be a hardware bug in some batches of C3 ([refer to reddit](https://www.reddit.com/r/esp32/comments/z0towa/comment/ixbo3a3/)). Espressif's docs state some [limitation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html#limitations):

> 4. data sent in the PC Terminal to ESP32-C3 direction (e.g. console commands), many PC Terminals will wait for the ESP32-C3 to ingest the bytes before allowing you to sending more data. This is in contrast to using a USB-to-Serial (UART) bridge chip, which will always ingest the bytes and send them to a (possibly not listening) ESP32-C3.

Since XIAO ESP32-C3 uses no additional USB-to-UART bridge and the native USB connection of the C3 is used, this might be the problem. Somehow it allows programming through this connection but I don't get Serial send to work (receiving works).

I'm okay closing this issue, since its not connected to any code issue on this project's code.

Thanks again fpr you help! :)

dchristl, 2023-08-09:
Ok sure, too bad it's a hardware issue, in that case, I can only close the issue. I've attached an early version with the mentioned portmap feature for you, maybe that will help you make progress for now. Please note, however, that everything is relatively untested and potentially buggy. But update with OTA should be possible, when the tested version is out.

[esp32c3nat_extended_full_v6.3.0-DEV.zip](https://github.com/dchristl/esp32_nat_router_extended/files/12303963/esp32c3nat_extended_full_v6.3.0-DEV.zip)
[esp32c3nat_extended_update_v6.3.0-DEV.zip](https://github.com/dchristl/esp32_nat_router_extended/files/12303965/esp32c3nat_extended_update_v6.3.0-DEV.zip)

Kind regards,
Danny

closed 2023-08-09 by dchristl

dchristl, 2023-08-09:
Thanks for your feedback. Happy to hear that it will work for your use case.  After testing, I probably intended to revise the UI a bit before taking screenshots anyway. Perhaps I'll also provide an explanation directly on the website, similar to how it's done for the other advanced settings.
If you come across any errors or if something doesn't work, I'd appreciate your feedback. This version also includes the latest version of the Espressif framework, which might break a few other things as well.

Best regards,
Danny