dchristl/esp32_nat_router_extended #76: No Wifi after uploading

richardwpaul, 2023-08-08:
A new ESP32. I followed your directions for uploading using windows graphic interface. Everything went as you describe in your directions with no errors. I rebooted the esp32. However there is no WiFi output from esp32.  Your suggestions, please.

dchristl, 2023-08-09:
Hello @richardwpaul ,

my first guess is, that you have either used the wrong version (full/upgrade) or the wrong address. You have to use the **full** version on address **0x0** on new devices. Have  you checked that and all the setings provided in the scressnshot? 
Otherwise please share a screenshot of your settings with the log (the cli window) included.

Regards,
Danny

richardwpaul, 2023-08-11:
Here are my steps: Load full, unplug to rebootNo wifiChange address and load update, unplug to rebootNo wifiLast image is from serial port when I reboot. Rick Paul765.430.5389*HP*gmail From: dchristlSent: Wednesday, August 9, 2023 1:04 AMTo: dchristl/esp32_nat_router_extendedCc: richardwpaul; MentionSubject: Re: [dchristl/esp32_nat_router_extended] No Wifi after uploading (Issue #76) Hello @richardwpaul ,my first guess is, that you have either used the wrong version (full/upgrade) or the wrong address. You have to use the full version on address 0x0 on new devices. Have you checked that and all the setings provided in the scressnshot?Otherwise please share a screenshot of your settings with the log (the cli window) included.Regards,Danny

richardwpaul, 2023-08-11:
Danny, Here is what I did. Erased flash, uploaded, rebooted. Still no WIFI Rick Paul765.430.5389*HP*gmail From: dchristlSent: Wednesday, August 9, 2023 1:04 AMTo: dchristl/esp32_nat_router_extendedCc: richardwpaul; MentionSubject: Re: [dchristl/esp32_nat_router_extended] No Wifi after uploading (Issue #76) Hello @richardwpaul ,my first guess is, that you have either used the wrong version (full/upgrade) or the wrong address. You have to use the full version on address 0x0 on new devices. Have you checked that and all the setings provided in the scressnshot?Otherwise please share a screenshot of your settings with the log (the cli window) included.Regards,Danny

richardwpaul, 2023-08-11:
Danny, First erased flash. Rick Paul765.430.5389*HP*gmail From: dchristlSent: Wednesday, August 9, 2023 1:04 AMTo: dchristl/esp32_nat_router_extendedCc: richardwpaul; MentionSubject: Re: [dchristl/esp32_nat_router_extended] No Wifi after uploading (Issue #76) Hello @richardwpaul ,my first guess is, that you have either used the wrong version (full/upgrade) or the wrong address. You have to use the full version on address 0x0 on new devices. Have you checked that and all the setings provided in the scressnshot?Otherwise please share a screenshot of your settings with the log (the cli window) included.Regards,Danny

richardwpaul, 2023-08-11:
Sorry for the email mess. 
[image]
[image]

dchristl, 2023-08-11:
Have you changed the default baud rate? It says 576000 and not 115200. It is not visible in your screenshot (under COM). Please change it and flash again. Some tools are "special" when the folder contains spaces. So please put the download tool and the bin in an folder without it. 
If this will not work either, try to [establish a serial connection](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html#windows-and-linux) with putty and post the log.

richardwpaul, 2023-08-11:
Sorry for the email mess.

richardwpaul, 2023-08-11:
1. erased using esp_tool erase_flash. 2. moved Bin file to Bin directory under flash_download_tool directory. 3. Flashed FULL using 0x0 and 576000. no errors. Image of serial port when booting. Sorry for the hassle.
[image]

dchristl, 2023-08-12:
> 2. moved Bin file to Bin directory under flash_download_tool directory

Your flash download tool directory seems to have a space, too: C:\Users\Rick Paul" ?

> 3. Flashed FULL using 0x0 and 576000. 

Please use **115200** like I mentioned. The wrong baud rate can cause invalid flashing results.

> Invalid header 

Invalid means corrupt image, mostly non exxistent bootloader. Can you try to download the image again?

closed 2023-12-15 by dchristl