dchristl/esp32_nat_router_extended #78: wifi ESSID with spaces not correctly managed

ulipo, 2023-08-20:
If the uplink wifi ESSID contains spaces, only the first word is kept in memory and if you pus "Apply", the wrong ESSID is saved.

dchristl, 2023-08-21:
Hello @ulipo 

I've tried with both SSIDs and it works like expected:

[image]

I can edit and connect to uplink without any issues. How do you mange to input the data? Do you have the latest version installed? 

Kind regards,
Danny

ulipo, 2023-08-21:
I'm using version 6.3.0, my ESSID is "Installazione App..", I can enter it
normally and it works, but when I go back to the interface it shows only
"Installazione" and if I hit Apply for any reason, it changes to that ESSID.

ulipo, 2023-08-21:
> Same behaviour, no problem:
> [image: Screenshot_20230821-141515_Fennec]
> <https://user-images.githubusercontent.com/2836933/262020100-74fe494c-49a3-404f-b776-3dc37aa4efc0.png>
>
> Besides the fact, that using of whitespaces and special characters should
> be avoided in SSIDs, can you describe exactly what you are doing? And what
> do you mean by going back? Does the connection work at all?
>
> The workflow should be:
>
>    1. Connect with a device to the ESP32-NAT-Router-SSID
>    (ESP32_NAT_Router)
>    2. Input your Uplink network (Installazione App..) and password
>    3. Apply
>    4. Connect to ESP32-NAT-Router-SSID or upstrema ID and check the inputs

I do exactly as you stated.
By "going back" to the interface I mean reconnecting at a later time to the
ESP32NAT web interface to check for updates and so on...

dchristl, 2023-08-22:
Does the connection work in general? Can you try another browser or which one do you use? I can't fix anything, if I can't reproduce the problem.

ulipo, 2023-08-23:
The connection works like a charm!
I tried with both Firefox and Chrome on two different Linux machines
(Debian and Ubuntu) and Firefox on Android, same results.
The only thing that's common to all this systems is the italian locale, I
tried to fake a different locale on Firefox/Linux with an extension, but
didn't change a thing.

dchristl, 2023-08-23:
So, everyting is saved correctly and it is only a view problem. The locale shouldn't be a problem. You can also try _wget_ if you are on Linux. I've also used Firefox, Chrome on Ubuntu and Fennec (Firefox-clone) on Android and everything seems to work fine.
Do you have another chip to try?  Can you open a terminal when visiting the site? Is there anything to see here?

ulipo, 2023-08-24:
Unfortunaltely I have no other chip at the moment.
I connected the board to my PC and [dumped the serial output](https://github.com/dchristl/esp32_nat_router_extended/files/12429605/terminal.txt), in sequence that's what I did:
- booted the board
- opened configuration page
- applied new parameters (with the correct SSID)
- waited for board rebooting

- reloaded config page

dchristl, 2023-08-24:
I can't see anything suspicious here. The only thing is, that you use a dev version. Could you update to the latest release version instead?

ulipo, 2023-08-24:
That's quite strange, I flashed manually 6.0.0 an then used OTA to update, it should have been the release version...
When I'll find some spare time, I'll try to completely erase the device and flash anew.

dchristl, 2023-08-26:
I apologize for the confusion. You have/had the correct version. There seems to be an error in reading the version number in the console. Nevertheless, my hands are tied here. I believe it might be an issue with the chip, but I'm not entirely sure. Since I can't replicate this, it's unfortunately unsolvable for me.

Two more options come to mind. Was there a past version where it worked? If so, which one? In that case, I could potentially check the repository to see if anything has changed at this part. Alternatively, you'd need to provide me with your inputs exactly, including the password, etc. It could be the combination of everything that is causing this issue.

ulipo, 2023-08-26:
I got myself a new esp32 board (not perfectly suitable as it only has an integrated antenna) and tried to install some versions.
Well, it seems that the problem started with v5.0.0, with v4.0.0 it works ok:
[image]

Here is the terminal dump:
[terminal.txt](https://github.com/dchristl/esp32_nat_router_extended/files/12446125/terminal.txt)

dchristl, 2023-08-27:
I think I've found the issue. The problem is the HTML file minification. The delivered HTML is heavily compressed to save space and reduce the amount of data transferred. This minifier removes duplicate double quotes in the HTML, which is why everything after the space is still in the HTML but not displayed anymore. When I'm developing, this feature is not active, which is why it worked for me. The latest version should solve your problem. You are able to update tis with OTA. Thank you for reporting the issue.

Kind regards,
Danny

ulipo, 2023-08-27:
Thanks Danny, I can confirm v6.3.1 it's working as intended.

dchristl, 2023-08-27:
Great, I will close this issue now.

closed 2023-08-27 by dchristl