dchristl/esp32_nat_router_extended #79: Feature request - WiFi to Ethernet Bridge

Radek-2, 2023-08-22:
Hello,
any chance to use this firmware with the **WT32-ETH01 module to make a WiFi to Ethernet Bridge ?** Or a combination of ESP32 + LAN8720 network module?

LAN8720 module [https://tinyurl.com/3zbjuzx3](url)
WT32-ETH01 [https://tinyurl.com/2czxxwws](url)
WT32-ETH01 datasheet [https://tinyurl.com/4eazy4sf](url)

In any case, great work and I wish you the best of luck with further development!

dchristl, 2023-08-22:
Hello, 

I'm not familiar with the module and its usage. Could you briefly explain what you intend to do with it? In your scenario, how would the router be connected? Through LAN or via WiFi?

If you only want to use the ESP32 without LAN, it should work like that as well. Utilizing the LAN would require a significant amount of implementation.

Kind regards,
Danny

jano1276, 2023-11-19:
Hello,
here is my scenario:

local router ---(ethernet 192.168.1.x)---> wt32-eth01 ---(wifi network 192.168.4.x)--->

regards,
Jan

dchristl, 2023-11-20:
Hello @jano1276 ,

Ok, I thought about something like this, but this makes not really a sense, because it will be incredible slow. Why not using ther Router's Wifi? 
Sorry, but I will not implement this, because it will be a complete rewrite of the code for some rare use cases, where better options are available.

Kind regards,
Danny

yingpwu, 2023-11-24:
In my scenario ,Router --->via WIFI--->ESP32 ---> via Ethernet  RJ45 ---->Device ,that would useful for a lot things ,many old stuff not support wireless ,that many will be a good project. 
Thanks for your project.

closed 2023-11-24 by dchristl