dchristl/esp32_nat_router_extended #80: Feature request - static IP based on mac address

jackyaway, 2023-08-27:
Would be great to have the ability to assign always the same IP address at devices, based on their mac. Otherwise, if it's possible, to let only the main router to do dhcp (the result will be only a LAN 192.168.1.x and not a new LAN for every esp32 - 192.168.4.x, 192.168.5.x, etc.). Thank you!

dchristl, 2023-08-27:
Hello @jackyaway,

> Would be great to have the ability to assign always the same IP address at devices, based on their mac

the assignment of the same IP based on the MAC is already on my bucket list ;) But it takes some time.

> Otherwise, if it's possible, to let only the main router to do dhcp (the result will be only a LAN 192.168.1.x and not a new LAN for every esp32 - 192.168.4.x, 192.168.5.x, etc.).

This is already possible, but you have to manually set up a route and you have to disable NAT in the router. Have a look at [this issue](https://github.com/dchristl/esp32_nat_router_extended/issues/73) or [this explanation](https://community.home-assistant.io/t/declare-a-static-route-to-reach-other-lan/311442), how to set it up.

Kind regards,
Danny

jackyaway, 2023-08-27:
> This is already possible, but you have to manually set up a route and you have to disable NAT in the router. Have a look at [this [...]

Thank you for the reply! I have disabled nat in esp32 already and set a route to 192.168.4.0/24 via 192.168.1.240 in the main router (openwrt) but i can't get the devices to connect with 192.168.1.x IP address, only 192.168.4.x

dchristl, 2023-08-27:
I'm sorry, but I can't be of much help with that. I'm not an expert in that area and I don't have a router that supports that. Perhaps you could reach out to the issue creator in the linked issue or seek assistance from the HomeAssistant community to make progress.

jackyaway, 2023-08-27:
Thank you. I'll do some other tentative.

closed 2023-09-03 by jackyaway