dchristl/esp32_nat_router_extended #81: Slow WiFi speed after 2-3 Hours

mdmarufmkh, 2023-09-02:
I'm facing a problem where the download speed becomes very slow (under 100 KB/s) after being on for 2-3 hours. The upload speed is fine. The only way to fix this is a restart. Could you get me solution please.

dchristl, 2023-09-04:
Hello @mdmarufmkh ,

can you provide a log output when the error occurs? I can't really understand why the error is occurring without it, and especially only during the download. Perhaps you could also post the output of "free" afterwards. Thank you.

Kind regards,
Danny

mdmarufmkh, 2023-09-06:
I think I fixed the issue. I soldered an external antenna to my ESP32 for greater range. It did give me much more range but the chip was overheating and after some time the internet speed slows drastically. Then I set the transmission power to "Medium". It seems to fix the overheating and it now works fine. Thank you for your response.

closed 2023-09-06 by mdmarufmkh