dchristl/esp32_nat_router_extended #82: Unexpected LED behaviour

Kobbe1, 2023-09-10:
Hi,
first of all great project, was easy to setuip and works very well so far.
Bellow is no big deal, more of a note really.

The LED on my board (some generic ESP32 from China) does not seem to blink at all, it only lights up solidly indicating a connection to the upstream network, also with 2 clients connected downstream as seen on the web interface. Or maybe I just can't see it blink as itz is too fast?

If I disable it through web interface it is off as expected.

Also the description of the expected behabviour is not in your's fork readme, but only in the "original" https://github.com/martin-ger/esp32_nat_router? Or maybe I just did not see it ... 

Thanks a lot!

-T

dchristl, 2023-09-10:
Hello @Kobbe1,

thank you for your kind words.
The problem with the multitude of different boards is that they use different GPIOs, some are not even available, or are used for other purposes. Adapting this for all the boards would require a significant amount of work and extensive testing across many different boards. I believe I can better utilize my limited time for this project elsewhere. Therefore, I have decided not to describe/promote this feature. I hope you understand.

Kind regards,
Danny

Kobbe1, 2023-09-10:
Thanks. The feature is not important for me, functionality before aesthetics. Well for me anyway :) So totally agree with what you are saying on priorities.
The webserver description still has the feature so I thought it is supported/still implemented. The GPIO used is obviously correct as it switches off and on with the switch in the screenshot, just no blinking light show ;)

I think we can close that issue unless you want more inputs ...

[image]

dchristl, 2023-09-11:
Yes, that is still in the interface because I carried it over from the old project anyway. This way, it can be deactivated if the blinking is annoying, the GPIO is used for something else, or the blinking behavior is faulty. ;) 
It's somewhat strange that the deactivation works (meaning the correct GPIO is being addressed), yet the blinking is still faulty. I suspect there might be an issue with the board itself, but it will be generally difficult to reproduce. To do so, I would need the exact same board as you have. Therefore, I would suggest closing the issue for now and, if I have a lot of free time in the future, reopening it.

closed 2023-09-11 by dchristl