dchristl/esp32_nat_router_extended #84: Feature request : Add WEP support on STA

Dh0mp5eur, 2023-09-10:
Hello, Thank you for your great project.

It's possible to add WEP support for STA mode?
This would be very useful for old network equipment.

Thanks

dchristl, 2023-09-11:
Hello @Dh0mp5eur ,

it is possible in general, of course and not so hard to implement. I'll have to see when I can get to the implementation, so I can't promise a date yet.

Kind regards,
Danny

dchristl, 2023-09-24:
Sadly, this can't be done, because this is not supported anymore. The [documentations](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_ap_config_t) says:

> Auth mode of soft-AP. Do not support AUTH_WEP, AUTH_WAPI_PSK and AUTH_OWE in soft-AP mode. When the auth mode is set to WPA2_PSK, WPA2_WPA3_PSK or WPA3_PSK, the pairwise cipher will be overwritten with WIFI_CIPHER_TYPE_CCMP.

So WEP is not supported anymore. You have to use an open network instead. 

Regards,
Danny

closed 2023-09-24 by dchristl