dchristl/esp32_nat_router_extended #85: compile error

epowsal, 2023-09-17:
[855/857] Linking CXX executable esp32_nat_router_extended.elfFAILED: esp32_nat_router_extended.elf bj
cmd.exe /C "cd . && M:\work\tool\Espressif_ide2.9.0\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\esp32_nat_router_extended.elf.rsp -o esp32_nat_router_extended.elf  && cd ."
m:/work/tool/espressif_ide2.9.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj):(.literal.main_task+0x18): undefined reference to `app_main'
m:/work/tool/espressif_ide2.9.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(port_common.c.obj): in function `main_task':
M:/work/tool/Espressif_ide2.9.0/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/port_common.c:128: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the m:\work\tool\espressif_ide2.9.0\frameworks\esp-idf-v5.0.1\examples\esp32_nat_router_extended-master\build\log\idf_py_stderr_output_12452 and m:\work\tool\espressif_ide2.9.0\frameworks\esp-idf-v5.0.1\examples\esp32_nat_router_extended-master\build\log\idf_py_stdout_output_12452

animevietsub, 2023-09-17:
You are trying to make the project work in c++ right? 
In file `esp32_nat_router.cpp` you missing `extern "C"` in 
```c++
extern "C" void app_main(void)
{
...
}
```

dchristl, 2023-09-17:
Just read the [documentation how to build](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md) and use platformio

epowsal, 2023-09-22:
thanks, now it can compile by the build doc.

closed 2023-09-22 by epowsal