dchristl/esp32_nat_router_extended #88: Features request: customize AP netmask

zhaqianyuan, 2023-09-22:
I hope to be able to customize the AP netmask instead of selecting Class A, Class B, Class C，such as 255.255.255.240

dchristl, 2023-09-22:
Hello @zhaqianyuan,

What is the use case for this? Are you looking to create subnets? What would be the next steps once the feature is implemented?

Kind regards,
Danny

zhaqianyuan, 2023-09-23:
> Hello @zhaqianyuan,
> 
> What is the use case for this? Are you looking to create subnets? What would be the next steps once the feature is implemented?
> 
> Kind regards,
> Danny

yes. Because I don't enable NAT, I add a static route on the main router so that all hosts can be accessed from the main router network. This is usually a small-scale coverage. Each esp does not have many hosts. Class C address is too much for it. For example, I use 192.168.1.0/24 as the main router network, one esp use 192.168.2.0/28, It can connect 14 hosts，the second esp use 192.168.2.16/28, also have 14 hosts

dchristl, 2023-09-23:
Thank you for the explanation. I need to dive deeper into the topic before I can implement it, but I've understood your plan. Unfortunately, I currently lack the time to implement it in the near future. I will see when I can fit it in.

dchristl, 2023-09-23:
I didn't mention it earlier because I hadn't tested it yet. However, I have now, and it works. Perhaps that's sufficient for you. It's already possible to change the subnet mask to a different value, just not through the GUI. To do that, you need to establish a serial connection and enter the following: 

```shell
nvs_namespace esp32_nat
nvs_set netmask str -v "255.255.255.240"
restart
```

After that, the subnet mask will also be assigned to the clients in that way. In the UI, it still looks slightly strange, but I think that can be overlooked:

[image]

My connected devices are getting the correct subnet mask.

closed 2023-10-02 by dchristl

dchristl, 2023-10-02:
I opted for the feature as it was easy to implement. Starting from version 6.4.0, the netmask can be set in UI. Feedback is welcome. Enjoy!

commit 24a222d7 (2023-09-24): Fully implemented option to set custom IP. closes #88