dchristl/esp32_nat_router_extended #89: Feature request: Channel width selection (20Mhz or 40Mhz)

mdmarufmkh, 2023-09-23:
Hi, love your work. Thank you so much for this project. I'm using this on my home on a daily basis. Just here to say is it possible to add a channel width selection feature to this project? Because, in my home my nat router works really well. But in my work place where 2.4Ghz is a warzone it doesn't seem to function properly. So it'll be very nice if I could just make the channel width 20Mhz to reduce interference from other networks.

dchristl, 2023-09-23:
Hello @mdmarufmkh ,

thank you for your kind words. The feature is a great idea, and I'll have to see when I can find the time for its implementation. Currently, I have many other things in the pipeline.

Kind regards,
Danny

dchristl, 2023-10-02:
I opted for the feature as it was easy to implement. Starting from version 6.4.0, the bandwidth can be switched. Feedback is welcome. Enjoy!

closed 2023-10-02 by dchristl

mdmarufmkh, 2023-10-03:
Thank you so much. Can't wait for the next update! 😊