dchristl/esp32_nat_router_extended #91: Can't connect to station, whet "$" in password

pascaltippelt, 2023-10-08:
Hi!

At first: Thank you for your work. It is stable and more reliable than other solutions. I really like it.

Unluckily, if i want wo connect to a WiFi with a "$" in the passowrd (example: "password$$"), I find myself in a "connect-loop":

```raw
I (163970) ESP32NRE: disconnected - retry to connect to the STA
I (163980) ESP32NRE: retry to connect to the STA
I (163980) wifi:primary chan differ, old=6, new=6, start CSA timer
I (163990) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (163990) wifi:state: init -> auth (b0)
I (164000) wifi:state: auth -> assoc (0)
I (164000) wifi:state: assoc -> run (10)
I (167010) wifi:state: run -> init (fc0)
I (167010) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (167010) ESP32NRE: disconnected - retry to connect to the STA
I (167030) ESP32NRE: retry to connect to the STA
W (169060) wifi:Haven't to connect to a suitable AP now!
I (169860) ESP32NRE: disconnected - retry to connect to the STA
I (169860) ESP32NRE: retry to connect to the STA
I (169870) wifi:primary chan differ, old=6, new=6, start CSA timer
I (169870) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (169880) wifi:state: init -> auth (b0)
I (169890) wifi:state: auth -> assoc (0)
I (169890) wifi:state: assoc -> run (10)
I (172900) wifi:state: run -> init (fc0)
I (172900) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (172910) ESP32NRE: disconnected - retry to connect to the STA
I (172920) ESP32NRE: retry to connect to the STA
W (175340) wifi:Haven't to connect to a suitable AP now!
I (175760) ESP32NRE: disconnected - retry to connect to the STA
I (175760) ESP32NRE: retry to connect to the STA
I (175770) wifi:primary chan differ, old=6, new=6, start CSA timer
I (175770) wifi:new:<6,2>, old:<6,2>, ap:<6,2>, sta:<6,0>, prof:1
I (175780) wifi:state: init -> auth (b0)
I (175780) wifi:state: auth -> assoc (0)
I (175790) wifi:state: assoc -> run (10)
I (178800) wifi:state: run -> init (fc0)
```

Using the same AP with another password, not containing a "$", everything works fine.

I am not shure how to fix this, but there might be an error handing over the password from the web gui. If it is not fixable, it should at least be mentioned!

I do know those ESP-Modules can connect to this WiFi with the "$" in the password, as it works with arduino, esphome and so on.

Best

Pascal

dchristl, 2023-10-09:
Hi @pascaltippelt ,

thans for your bug report. I will have a look into this.

Best regards,
Danny

dchristl, 2023-10-09:
I've tested this out and it works like expected. The password with $ is saved correctly and connection works.  I have checked password in AP (like you mentioned) and also in STA.

Do you have the latest version installed? You mentioned problems with AP, but your log shows failure in connection to STA.

pascaltippelt, 2023-10-10:
Hi @dchristl 
I tried to reproduce my problem with an other esp module. As it seems I had a defective module. It caused trouble somewhere else too. You are right, it works as expected. I'm sorry I disturbed you.

closed 2023-10-10 by pascaltippelt