dchristl/esp32_nat_router_extended #92: Feature Request: OLED status screen

nightflasher, 2023-10-26:
I wonder, if it is possible to add an i2c OLED screen to show current connected devices, maybe the device hostname too?

dchristl, 2023-11-16:
Hello @nightflasher .

of course this will be possible, but I'm afraid it will not fit in really well into this project. An extra display will slow down the device a bit (for prcessing output) and will also cost a lot of power. This also contradicts somewhat with the small energy-saving access point that lies invisibly in the corner and amplifies/extends the Wi-Fi. Currently, I am planning to implement a REST API that could possibly fetch the required information; then, it could be implemented through a separate server for your own.
BTW: I would stick more to an e-ink-Display than an OLED. 

Sincerely,
Danny

tarikjmr, 2023-12-13:
But an oled display will give it a good feature. Please modify the code with 12c oled :-)

dchristl, 2023-12-15:
Sorry, I think everything is said. I will close this issue now. You are free to fork this project and implement your requested feature. 

Thank you

closed 2023-12-15 by dchristl