dchristl/esp32_nat_router_extended #95: Future request: Serial over Wifi

xiv3r, 2023-11-30:
E.G
[image]
[image]

dchristl, 2023-12-01:
Hello @xiv3r ,

I have been planning a feature of this kind for a while, but not through a GUI, rather through a web service (e.g., using curl). Essentially, an interface where all the features available via the serial connection can also be executed through a web request. Unfortunately, I haven't had the time to implement anything like that so far.

Regards,
Danny

xiv3r, 2023-12-24:
can you add STA Static IP
[image]

xiv3r, 2023-12-24:
Can you add Station Static IP settings.. thanks!

[image]