dchristl/esp32_nat_router_extended #96: slower than Martin's code

piotrcurious, 2023-12-08:
I tried both versions, the original of the Martin Ger
and this one
And this one works much slower (9mbit vs only 3mbit), regardless of bandwidth settings etc, at exactly same location etc.
I use precompiled binary and standard esp32 dual core 240mhz

dchristl, 2023-12-08:
Hello @piotrcurious ,

I don't really have an explanation for why martin-ger's code should be faster. The code for the routing itself is unchanged (aside from always using the latest version of the framework). My modifications mostly focus on UI and functionality extensions. I haven't been able to observe any performance differences. Do you perhaps have multiple different ESP32 devices to validate this?

Kind regards,
Danny

piotrcurious, 2023-12-08:
i have few identical ESP32 boards (WROOM-32D) on which i did  test your "extended" version and also "plus" version which both perform slower than original Martin Ger version.
I used precompiled binaries in each case so might be they are not optimized correctly? It seems Martin Ger code performance depends heavily on optimizations used. 
Might be that esp-idf changed in the meantime too, just slight code size changes could trigger cache misses. 
I really have no idea and right now i have no way of doing much more research. 
I tested everything close to the upstream router. 

Greetings.

dchristl, 2023-12-09:
> I used precompiled binaries in each case so might be they are not optimized correctly? It seems Martin Ger code performance depends heavily on optimizations used.

Can you please elaborate on this? I can't find any clues to it, neither in the code ([Makefile ](https://github.com/martin-ger/esp32_nat_router/blob/master/Makefile)) nor in the [documentation](https://github.com/martin-ger/esp32_nat_router#building-the-binaries-method-2---platformio). The build with PlatformIO is even identical to my approach. So no compiler flags or something else. 
In the meantime, martin-ger is also compiling against ESP-IDF 5.1, like this project. I will perform speed tests again with both versions and see, but currently, I cannot find any justification for the differences.

piotrcurious, 2023-12-09:
Hum well yes. Makefile does not define ANY default optimization flags.
Martin states in readme he uses Og or Os for testing, which means some flags have influence, but he did not investigate which... default optimization flags change from gcc to gcc versions too.
You need to set CFLAGS manually prior each compile as environment variable, they are not defined

-------- Original Message --------

piotrcurious, 2023-12-09:
and this means CFLAGS and CXXFLAGS as it's c++ and preprocessor is gcc

-------- Original Message --------

dchristl, 2023-12-10:
I have now tested both projects 10 times using iPerf and come up with exactly the same results (average of 13.5 MBit/s). These are also the values specified in Martin's project. The documented optimizations (0g, 0s) disable default optimizations, see [documentation](https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html):

_-Og_
_Optimize debugging experience. -Og should be the optimization level of choice for the standard edit-compile-debug cycle, offering a reasonable level of optimization while maintaining fast compilation and a good debugging experience. It is a better choice than -O0 for producing debuggable code because some compiler passes that collect debug information are disabled at -O0._

_Like -O0, -Og completely disables a number of optimization passes so that individual options controlling them have no effect. Otherwise -Og enables all -O1 optimization flags except for those that may interfere with debugging:_

_-Os_
_Optimize for size. -Os enables all -O2 optimizations except those that often increase code size:_

After all, I can see no difference, the only one is the esp-idf-Version (my is 5.1.1 martin's is 5.1.2). I don't think the difference in patch level is the cause for your slower device. Sorry, but I don't see any chance to help you here.

piotrcurious, 2023-12-10:
Do not worry, i can just use Martin Ger's version.
All those projects helped me to have internet in a place i otherwise could not so did their job :)
I report merely because i spotted there is a measurable difference in speed so i guess real reason will surface out sooner or later.
I did speed testing by using speedtest.pl .
Results are either
9mbit for Martin Ger
3-4mbit for "enhanced" version
3-4mbit for "plus" version.
Web browsing is also noticeably slower, often having "stalls" when the page just waits for a second or two until it loads.

speedtest.pl opens multiple connections to various servers to make the test more statistically relevant for typical web browsing experience, so perhaps the problem is that Martin Ger's version can handle more concurrent connections?
Perhaps a problem then is available free memory?

Sorry i did not review the code through, i have very little time. I am happy i could set up the repeater from something out of the junk box and i have to move on doing something else.
Greetings and thank You all for the great work!
-------- Original Message --------

dchristl, 2023-12-10:
> I did speed testing by using speedtest.pl .
Results are either
9mbit for Martin Ger
3-4mbit for "enhanced" version
3-4mbit for "plus" version.

I think these values are MByte and not MBit, right? I have with both versions about 3,5 MB/ s download. Are you sure, you were connected to the ESP32 router on your first test. The device is not capable of such speed and martin wrote in the readme much slower speed, too (16MBit/s = 2MB/s )

The current development-version of this project is now compiled against ESP-IDF 5.1.2 (same as martin's version). You can try it out, if you switch to the canary channel. [Connnect with terminal](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/establish-serial-connection.html), followed by thes two commands

`nvs_namespace  esp32_nat`
`nvs_set canary i32 -v 1`

The OTA-Update will show you, that you are on canary channel:
[image]

You can now "Install the latest version".

Please let me know, if there are any improvements.

piotrcurious, 2023-12-10:
screenshots from the test (connected to Martin Ger extender)
So those are mbit speeds i guess. it's much lower than 16mbit as it's normal speed test so there is burden of ack tcp packets going in opposite side.
The upstream router is 200mbit fiber optic, so i would notice if i would be connected upstream ;)
-------- Original Message --------

dchristl, 2023-12-10:
OK, 9 MBit is possible, without a doubt, but really slow. Can you please update, like I mentioned. Because I have seen a (small) performance improvement on my devices, too. So, maybe the framework had some flaws there and the new version will fix some.

dchristl, 2023-12-15:
I've released a new version, compiled agaunst esp idf 5.1.2. So no need to connect the device by serial and input the commands. You can just do a normal OTA-Update.

closed 2024-01-19 by dchristl