dchristl/esp32_nat_router_extended #97: Repeater function

mkfrcp, 2023-12-30:
Hello,
Very nice UI!

Can you show me how to use your esp32_nat_router as a repeater instead of an extender?
I just want to extend the range of my wifi without changing wifi settings of the clients.

Thank you so much!

dchristl, 2023-12-30:
Hello @mkfrcp ,
the repeater function is not like a real repeater or router, as this mode is not supported by the ESP32. However, you can simulate this by assigning the same credentials to the ESP32 network as those for the uplink network. The devices will then connect to the strongest signal.

Kind regards,
Danny

closed 2024-01-19 by dchristl

mkfrcp, 2024-01-19:
> <dchristl/esp32_nat_router_extended/issue/97/issue_event/11539490032@
> github.com>

Hello.

Thanks a lot for your help !

Regards,