dchristl/esp32_nat_router_extended #98: More than two clients?

Schorschlhuber, 2024-01-01:
Hello,
thank's for your work and a happy new year! I've tried to connect more than two devices but the DHCP does only provide 2 adresses, so two devices share the IP 192.168.4.2 or 192.168.4.3. 
 

Does the router only support 2 Clients?

dchristl, 2024-01-01:
Hello @Schorschlhuber ,

wish you a Happy New Year, too. The router supports up to 255 clients as default. I think there will be some performance issues if all possible cliebnts will connect, but 10 devices should be no problem.

Kind Regards,
Danny

Schorschlhuber, 2024-01-01:
Very strange. The problem persists not only with two Tasmota-devices but also with my smartphone. All three devices get IP .3 and kick each other out. "Connected devices" shows three devices with two devices IP 0.0.0.0

After disconnect the two other devices for about an hour, the devices are getting new adresses .5 and .8.

Kind Regards,
Schorschlhuber

Schorschlhuber, 2024-01-02:
The problem sill exists even with other devices. It seems the DHCP offers the same IP do multiple devices. How long is the DHCP-Lease-Time? If the third devices connects, another will kicked out. 

[image]

dchristl, 2024-01-04:
That is really very strange. Perhaps there have been changes in functionality due to the framework update (esp-idf), causing it not to work anymore. The default DHCP server from the Soft-AP is used here, which I use as is without modifications. I will take a closer look, but unfortunately, I have little time for this project right now. If I remember correctly, the default lease time was 2 hours

Schorschlhuber, 2024-01-04:
My workaround: set static IP per client > 172.16.4.10 That works without any problems.

dchristl, 2024-01-08:
I've tried the latest version and have no problem to connect multiple devices. I've tested with an ESP32. The clients were 3 Smartphones and one notebook:

[image]

[image]

The [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html) says that the maximum number of connections is 15 at max for the ESP32 and the ESP32 S2. The ESP32 C3 is limited to 10. 

I unfortunately cannot reproduce the issue. Which chip are you using? Do you have another ESP32 for testing? Could you possibly connect via the serial connection and post the log when the third client attempts to connect?

dchristl, 2024-01-19:
Please reopen, if the issue persist, but currently I can't do anything about it.

closed 2024-01-19 by dchristl