dchristl/esp32_nat_router_extended #99: Trying to build --- cmd_router.c  IP_NAPT must be defined

candrews67, 2024-01-04:
I have attempted to build for several different chips but no matter what I do I get the same error message - cmd_router.c:26:2: error: #error "IP_NAPT must be defined".  Can someone point me in the right direction to fix this?  I have installed the sdk on a Linux computer and a MAC but get the error on both.  

Thanks!

dchristl, 2024-01-05:
Hello @candrews67 ,

have you looked at the [How to build section](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md) There is everything described. Your problem seems to be, that you don't use platformio and your settings (sdkconfig) fall back to default. So all the[ config changes](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/advanced.md#modified-parameters-compared-to-the-default-configuration), including enabled NAT (your error) is simply not available.

Regards,
Danny

candrews67, 2024-01-06:
Thanks!!

After I posted I found where others had had similar issues so I have started setting up platformio, that is an entirely new endeavor.  I have an Ubuntu 23.10 install and platformio is not impressed.

Based on what I have seen in other closed tickets and your comment I am hoping once I get platformio working I will be able to successfully compile.

Thanks for your time and response.

candrews67, 2024-01-06:
Thanks. Working on setting up platformio

closed 2024-01-06 by candrews67