dchristl/esp32_nat_router_extended #101: Can't setup gpio wake-up on deepsleep.

binarylefmart, 2024-01-15:
Hi,
Firstly, thanks for your fork, the code have been refresh and been really pleasant to use.
Secondly, I have an issue to setup the deepsleep on my wroom32.
I wish to always have the device on deepsleep mode and when my light sensor is up, wake up the device and keep the state up until the night come (and the state sensor down).
My principal issue is when I try to setup the gpio wake up, the called function is visibly a WiFi wake up and not a gpio wake up (and said than the esp32 is not compatible so).
It is a bug ?
And my second question is: it is possible to set the deepsleep state automatically at boot and after the state sensor of the wake up gpio is down ?
Thanks for your time, your reply and your work :) .

binarylefmart, 2024-01-17:
Hi dude,
Thanks for your feedback.
This could be a very nice feature to implement, I will follow the project closely ^^.
During this time I will put a 20w solar pannel and an 1s3p liion lol.
Thanks

dchristl, 2024-01-20:
So, I've researched a bit. 

> My principal issue is when I try to setup the gpio wake up, the called function is visibly a WiFi wake up and not a gpio wake up (and said than the esp32 is not compatible so).

You are correct in your statement about Wifi-Wakeup. This was implemented as such in the forked code by martin-ger. I believe this part should be fixed now in the current canary version of this project. If you want to try it out, you can temporarily switch it in the OTA update via serial console:

```
nvs_namespace esp32_nat
nvs_set canary i32 -v 1
```

You can then use the OTA update to install the version:
[image]

Keep in mind that the version may still contain errors. For example, error handling has not been adjusted yet. I would appreciate your feedback.

binarylefmart, 2024-01-22:
Hi Chris,
So, I tested the latest dev canary and the deep gpio wakeup and... It's work ! the gpio wakeup is fixed and call the right function.
Thanks for the search and the solution :)
So now I need to found how to setup constantly the deep sleep until the gpio state is to 0 (for the wake up) and turnedback to deepsleep when the gpio state is on 1.
If you have any Idea I take it :)
Thanks dude !

dchristl, 2024-01-22:
Great to hear that. The error was already present when I forked the project from martin-ger. ;)

For your other problem, unfortunately, I don't have a solution. I would probably approach it using a controllable power outlet (if the sensor supports REST, perhaps?), a Raspberry Pi with something like openHAB, or another Wi-Fi-enabled microcontroller (e.g., ESP32). I can't think of a solution that doesn't involve an additional device. I would opt for something related to power rather than trying to solve it with sleep; I think that would be simpler.