dchristl/esp32_nat_router_extended #105: ESP32-C6 ?

ErfanDL, 2024-02-03:
Hi @dchristl 
It would be great if you could develop the ESP32-C6 version so we can use 802.11AX Wi-Fi 6

dchristl, 2024-02-09:
Hello @ErfanDL ,

I don't think adding this boar should be a real problem. But the testing is. If you have such a board and want to test I'm happy to provide binarys for flashing. If everything works fine, I can add it to the releases.

Kind Regards,
Danny

ErfanDL, 2024-02-10:
> Hello @ErfanDL ,
> 
> I don't think adding this boar should be a real problem. But the testing is. If you have such a board and want to test I'm happy to provide binarys for flashing. If everything works fine, I can add it to the releases.
> 
> Kind Regards, Danny

Thanks for your reply, I have built my own ESP32-C6 development board. I would be grateful if you could provide the ESP32-C6 binary for me to test.

[image]

dchristl, 2024-02-10:
Here we go:
[firmware-esp32-c6.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14228842/firmware-esp32-c6.zip)

You can flash the esp32-c6-nat_extended_full.bin like any other device or (if it doesn't work, because I can't test it) or the bootloader, partitionand firmware with 
`esptool.py  --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin
`

If everything works I can add it to the "official" releases and make OTA-updates available.

ErfanDL, 2024-02-10:
> Here we go: [firmware-esp32-c6.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14228842/firmware-esp32-c6.zip)
>
> You can flash the esp32-c6-nat_extended_full.bin like any other [...]

thanks for the binary. I uploaded the bins to ESP32C6 but it getting boot loop

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40018bea
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd04
load:0x4086e610,len:0x2cbc
load:0x71630008,len:0x17fd06e6
Invalid image block, can't boot.
ets_main.c 331 
```

dchristl, 2024-02-10:
Now it is hard for me to help you without a device. Maybe it is possible that [the bootloader is at address 0x0](https://github.com/espressif/esp-idf/issues/12281) instead of 0x1000.

You can change your command to:

`esptool.py  --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin `

ErfanDL, 2024-02-10:
I did this and it no longer bootloops, but the "ESP32_NAT_Router" ssid not displayed.

DevCode01, 2024-02-10:
I did this too but now my ESP32-C is not responding

ErfanDL, 2024-02-10:
> it's working, thank you

How did it work for you?

dchristl, 2024-02-11:
> it's working, thank you

I'm also interested in, what you have done to make it work.

kondakos, 2024-02-23:
> it's working, thank you
I'm   interested in.

kondakos, 2024-02-23:
[image]

kondakos, 2024-02-23:
The results are Exellent. Direct connection from 4m  to home WIFI 2.4 Mhz AP from 21 Mbps/4Mbps goes to 7Mbps/3Mbps  when connection  from 4m with the ESP32C6.

ErfanDL, 2024-02-24:
> It's working, updated python and esptool with "esptool --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin"
> 
> the esptool report
> 
> C:\Users\tk\Downloads\esp>esptool -p COM6 flash_id esptool.py v4.7.0 Serial port COM6 Connecting.... Detecting chip type... ESP32-C6 Chip is ESP32-C6 (QFN40) (revision v0.0) Features: WiFi 6, BT 5, IEEE802.15.4 Crystal is 40MHz MAC: 54:32:04:ff:fe:XX:XX:XX BASE MAC: 54:32:04:XX:XX:XX MAC_EXT: ff:fe Uploading stub... Running stub... Stub running... Manufacturer: 5e Device: 4017 Detected flash size: 8MB Hard resetting via RTS pin...
> 
> Thank you

**so why its not worked for me ?** :(

Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 40:4c:ca:ff:fe:49:59:e4
BASE MAC: 40:4c:ca:49:59:e4
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0016cfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 21312 bytes to 12912...
Wrote 21312 bytes (12912 compressed) at 0x00000000 in 0.3 seconds (effective 492.8 kbit/s)...
Hash of data verified.
Compressed 1427904 bytes to 734995...
Wrote 1427904 bytes (734995 compressed) at 0x00010000 in 8.6 seconds (effective 1328.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 113...
Wrote 3072 bytes (113 compressed) at 0x00008000 in 0.0 seconds (effective 517.4 kbit/s)...
Hash of data verified.

dchristl, 2024-02-24:
I don't know this makes any difference, but have you seen the output of your command:

`Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum`

It seems you have an encrypted bootloader. Maybe you have success with 

`esptool.py --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size keep 0x0 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin`

You can also try to erase the ESP32 before with:

`esptool.py erase_flash`

ErfanDL, 2024-02-24:
> I don't know this makes any difference, but have you seen the output of your command:
>
> `Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the [...]

**I tried but it didn't worked. I never encrypted the bootloader on my esp32c6 here the summary**

espefuse.py v4.7.0
Connecting...
Detecting chip type... ESP32-C6
BLOCK0          (                ) [0 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000
MAC_SPI_8M_0    (BLOCK1          ) [1 ] read_regs: ca4959e4 fffe404c 00000000 08000000 00000000 00000000
BLOCK_SYS_DATA  (BLOCK2          ) [2 ] read_regs: ed8b8278 72492f16 1d7f5659 3b205fab ec269f78 071043f0 3dc781b8 11311111
BLOCK_USR_DATA  (BLOCK3          ) [3 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY0      (BLOCK4          ) [4 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY1      (BLOCK5          ) [5 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY2      (BLOCK6          ) [6 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY3      (BLOCK7          ) [7 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY4      (BLOCK8          ) [8 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY5      (BLOCK9          ) [9 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_SYS_DATA2 (BLOCK10         ) [10] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE_RD_RS_ERR0_REG        0x00000000
EFUSE_RD_RS_ERR1_REG        0x00000000

EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
ACTIVE_HP_DBIAS (BLOCK1)                           Stores the active hp dbias                         = 0 R/W (0b00000)
ACTIVE_LP_DBIAS (BLOCK1)                           Stores the active lp dbias                         = 0 R/W (0b00000)
LSLP_HP_DBIAS (BLOCK1)                             Stores the lslp hp dbias                           = 0 R/W (0x0)
DSLP_LP_DBIAS (BLOCK1)                             Stores the dslp lp dbias                           = 0 R/W (0x0)
DBIAS_VOL_GAP (BLOCK1)                             Stores the hp and lp dbias vol gap                 = 0 R/W (0b00000)
TEMP_CALIB (BLOCK2)                                Temperature calibration data                       = -12.0 R/W (0b101111000)
OCODE (BLOCK2)                                     ADC OCode                                          = 79 R/W (0x4f)
ADC1_INIT_CODE_ATTEN0 (BLOCK2)                     ADC1 init code at atten0                           = -76 R/W (0b1000010011)
ADC1_INIT_CODE_ATTEN1 (BLOCK2)                     ADC1 init code at atten1                           = -116 R/W (0b1000011101)
ADC1_INIT_CODE_ATTEN2 (BLOCK2)                     ADC1 init code at atten2                           = -124 R/W (0b1000011111)
ADC1_INIT_CODE_ATTEN3 (BLOCK2)                     ADC1 init code at atten3                           = -128 R/W (0b1000100000)
ADC1_CAL_VOL_ATTEN0 (BLOCK2)                       ADC1 calibration voltage at atten0                 = 12 R/W (0b0000000011)
ADC1_CAL_VOL_ATTEN1 (BLOCK2)                       ADC1 calibration voltage at atten1                 = 220 R/W (0b0000110111)
ADC1_CAL_VOL_ATTEN2 (BLOCK2)                       ADC1 calibration voltage at atten2                 = -240 R/W (0b1000111100)
ADC1_CAL_VOL_ATTEN3 (BLOCK2)                       ADC1 calibration voltage at atten3                 = -492 R/W (0b1001111011)
ADC1_INIT_CODE_ATTEN0_CH0 (BLOCK2)                 ADC1 init code at atten0 ch0                       = 0 R/W (0x8)
ADC1_INIT_CODE_ATTEN0_CH1 (BLOCK2)                 ADC1 init code at atten0 ch1                       = 0 R/W (0x8)
ADC1_INIT_CODE_ATTEN0_CH2 (BLOCK2)                 ADC1 init code at atten0 ch2                       = 0 R/W (0x8)
ADC1_INIT_CODE_ATTEN0_CH3 (BLOCK2)                 ADC1 init code at atten0 ch3                       = 0 R/W (0x8)
ADC1_INIT_CODE_ATTEN0_CH4 (BLOCK2)                 ADC1 init code at atten0 ch4                       = 0 R/W (0x8)
ADC1_INIT_CODE_ATTEN0_CH5 (BLOCK2)                 ADC1 init code at atten0 ch5                       = -4 R/W (0x9)
ADC1_INIT_CODE_ATTEN0_CH6 (BLOCK2)                 ADC1 init code at atten0 ch6                       = 0 R/W (0x8)

Config fuses:
WR_DIS (BLOCK0)                                    Disable programming of individual eFuses           = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                                    Disable reading from BlOCK4-10                     = 0 R/W (0b0000000)
SWAP_UART_SDIO_EN (BLOCK0)                         Represents whether pad of uart and sdio is swapped = False R/W (0b0)
                                                    or not. 1: swapped. 0: not swapped
DIS_ICACHE (BLOCK0)                                Represents whether icache is disabled or enabled.  = False R/W (0b0)
                                                   1: disabled. 0: enabled
[... cut ...]
Vdd fuses:
VDD_SPI_AS_GPIO (BLOCK0)                           Represents whether vdd spi pin is functioned as gp = False R/W (0b0)
                                                   io. 1: functioned. 0: not functioned

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             Represents whether RTC watchdog timeout threshold  = 0 R/W (0b00)
                                                   is selected at startup. 1: selected. 0: not select
                                                   ed

ErfanDL, 2024-02-24:
@dchristl 
can you build it for 16MB ESP32C6 ?

kondakos, 2024-02-24:
The flash size (8 vs 16 ) maybe is the difference that produce the error.

kondakos, 2024-02-24:
> @dchristl can you build it for 16MB ESP32C6 ?

You can try flash with specific  argument for 16Mb flash size. 

Setting Flash Mode and Size[](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#setting-flash-mode-and-size)

You may also need to specify arguments for [flash mode and flash size](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flash-modes.html#flash-modes), if you wish to override the defaults. For example:

esptool.py --port /dev/ttyUSB0 write_flash --flash_mode qio --flash_size 32m 0x0 bootloader.bin 0x1000 my_app.bin

Since esptool v2.0, these options are not often needed as the default is to keep the flash mode and size from the .bin image file. See the [Flash Modes](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/flash-modes.html#flash-modes) section for more details.

dchristl, 2024-02-25:
> can you build it for 16MB ESP32C6 ?

It is only possible to build for a chip not a specific hardware of it. As kondakos mentioned, you can try different parameters during flashing, although that shouldn't be necessary. 

You also seem to have exactly the same chip as him (QFN40) with 40 pins, so I can't quite understand why it's not working for you. Do you possibly have a second ESP32 C6 to rule out a hardware defect?

If I understand correctly, the flashing process worked in general. Could you please post the log from the serial console again? Maybe that will provide a clue.

kondakos, 2024-02-25:
The board that flashed is:
ESP32 / ESP32-C6 WiFi+Bluetooth Internet Of Things Dual Type-C Development Board Core Board ESP32-C6-DevKit C N4R2
https://www.aliexpress.com/item/1005005664230988.html

At first the flash image was Tasmota 13.4.0.1(arduino30) by Theo Arends working just fine.
https://ota.tasmota.com/tasmota32/tasmota32c6cdc-arduino30.bin
ESP Chip Id	16776716 (ESP32-C6 v0.0)
Flash Chip Id	0x17405E (QIO)
Flash Size	8192 KB
Program Flash Size	8192 KB

Which reports. 
c:\Users\tk\Downloads\esp>esptool -p COM7 flash_id
esptool.py v4.7.0
Serial port COM7
Connecting....
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 54:32:04xxxxxxxxxxxx
BASE MAC: 54:32:04:xxxxxxxxxx
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...

Flash the image :
c:\Users\tk\Downloads\esp>esptool --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x0 bootloader.bin 0x10000 firmware.bin 0x8000 partitions.bin
esptool.py v4.7.0
Found 1 serial ports
Serial port COM7
Connecting....
Detecting chip type... ESP32-C6
Chip is ESP32-C6 (QFN40) (revision v0.0)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 54:32:04:ff:fe:0c:49:00
BASE MAC: 54:32:04:0c:49:00
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x0016cfff...
Flash will be erased from 0x00008000 to 0x00008fff...
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 21312 bytes to 12912...
Wrote 21312 bytes (12912 compressed) at 0x00000000 in 1.3 seconds (effective 133.8 kbit/s)...
Hash of data verified.
Compressed 1427904 bytes to 734995...
Wrote 1427904 bytes (734995 compressed) at 0x00010000 in 64.7 seconds (effective 176.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 113...
Wrote 3072 bytes (113 compressed) at 0x00008000 in 0.0 seconds (effective 562.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Without doing anything the wifi router is available and working fine.

kondakos, 2024-02-25:
> > Hello @ErfanDL ,
> > I don't think adding this boar should be a real problem. But the testing is. If you have such a board and want to test I'm happy to provide binarys for flashing. If everything works fine, I can add it to the releases.
> > Kind Regards, Danny
> 
> Thanks for your reply, I have built my own ESP32-C6 development board. I would be grateful if you could provide the ESP32-C6 binary for me to test.
> 
> ![photo_2024-02-10_12-42-27](https://private-user-images.githubusercontent.com/14868771/303826693-9733d26c-ec02-4000-97b6-34c55c0d89ac.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg4ODIzNDAsIm5iZiI6MTcwODg4MjA0MCwicGF0aCI6Ii8xNDg2ODc3MS8zMDM4MjY2OTMtOTczM2QyNmMtZWMwMi00MDAwLTk3YjYtMzRjNTVjMGQ4OWFjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI1VDE3MjcyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA4MTI0ZjY4YjYyYTAxODk1MmJlYTU0NGVhMDAxMjJiN2EwNGU3MTFiYjNjMGUxNDI4ZjBhODU4YWNiOGU0ZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Gu7OC1ghwWjnwhj1tvc41IpyzQl2UWlgq615DzHS2fM)

You have to try another one  
 

https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#insufficient-capacitance 

https://user-images.githubusercontent.com/205573/30140831-9da417a6-93ba-11e7-95c3-f422744967de.jpg

kondakos, 2024-02-25:
> Hello @ErfanDL ,
> 
> I don't think adding this boar should be a real problem. But the testing is. If you have such a board and want to test I'm happy to provide binarys for flashing. If everything works fine, I can add it to the releases.
> 
> Kind Regards, Danny

Please add the image for ESP32C6 to the releases and provide updates. 
Thank you.

ErfanDL, 2024-02-26:
> It is only possible to build for a chip not a specific hardware of it. As kondakos mentioned, you can try different parameters during flashing, although that shouldn't be necessary.
>
> You also seem [...]

I tried every possible flash parameters but didn't worked. here the serial output:

mode:DIO, clock div:2
load:0x4086c410,len:0xa84
load:0x4086e610,len:0x2468
load:0x40875728,len:0x4e8
entry 0x4086c410
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x3 (LP_SW_HPSYS),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001974a
SPIWP:0xee

ErfanDL, 2024-02-26:
Now worked with external power (not USB) because the ESP32C6 internal USB using GPIO12 and the ESP32C6 keep reset factory at bootup !

[image]

dchristl, 2024-02-26:
Great, that it's working now. Could one of you both test the following:

- Flash the attached full.bin again. I have now modified it, and it should be flashable directly.
- After making some settings, please flash the firmware.bin from my first post at address 0x10000 (this simulates an OTA update) and check if the settings are retained.

If everything works, I can include the image in the next release. There is still some work to be done (OTA updates, pin assignments, sleep mode, build, etc.) before I can release it. However, with the attached pre-release version, most things should work for now.

[esp32-c6-nat_extended_full.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14405865/esp32-c6-nat_extended_full.zip)

ErfanDL, 2024-02-26:
> Great, that it's working now. Could one of you both test the following:
>
>     * Flash the attached full.bin again. I have now modified it, and it should be flashable directly.
>
>     * After making [...]

Yes everything works... Thank you very much for your efforts

kondakos, 2024-02-26:
From tasmota firmware  the new attached full.bin flashed. 
Working. 
After making some settings,   flashed the firmware.bin from   first post at address 0x10000
Everything working
Thank  you

Both firmware  have a bug.
When you put Bandwith 40Mhz  the device  crashed and bootlooping .

ELF file SHA256: 7d3efeb99fd9872e

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd4c
load:0x4086e610,len:0x2dd0
load:0x40875728,len:0x17d0
entry 0x4086c410
I (26) boot: ESP-IDF 5.1.2 2nd stage bootloader
I (27) boot: compile time Feb 10 2024 11:02:35
I (28) boot: chip revision: v0.0
I (30) boot.esp32c6: SPI Speed      : 80MHz
I (34) boot.esp32c6: SPI Mode       : DIO
I (39) boot.esp32c6: SPI Flash Size : 4MB
I (44) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (68) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (75) boot:  2 ota_0            OTA app          00 10 00010000 00177000
I (82) boot:  3 ota_1            OTA app          00 11 00190000 00177000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=420e0020 size=6c1b8h (442808) map
I (283) esp_image: segment 1: paddr=0007c1e0 vaddr=40800000 size=03e38h ( 15928) load
I (292) esp_image: segment 2: paddr=00080020 vaddr=42000020 size=d64bch (877756) map
I (650) esp_image: segment 3: paddr=001564e4 vaddr=40803e38 size=12d00h ( 77056) load
I (687) esp_image: segment 4: paddr=001691ec vaddr=40816b40 size=0373ch ( 14140) load
I (695) esp_image: segment 5: paddr=0016c930 vaddr=50000000 size=00068h (   104) load
I (706) boot: Loaded app from partition at offset 0x10000
I (707) boot: Disabling RNG early entropy source...
I (718) cpu_start: Unicore app
I (719) cpu_start: Pro cpu up.
W (728) clk: esp_perip_clk_init() has not been implemented yet
I (735) cpu_start: Pro cpu start user code
I (735) cpu_start: cpu freq: 160000000 Hz
I (735) cpu_start: Application information:
I (738) cpu_start: Project name:     esp32_nat_router_extended
I (745) cpu_start: App version:      DEV-VERSION
I (750) cpu_start: Compile time:     Feb 10 2024 11:01:44
I (756) cpu_start: ELF file SHA256:  7d3efeb99fd9872e...
I (762) cpu_start: ESP-IDF:          5.1.2
I (767) cpu_start: Min chip rev:     v0.0
I (771) cpu_start: Max chip rev:     v0.99
I (776) cpu_start: Chip rev:         v0.0
I (781) heap_init: Initializing. RAM available for dynamic allocation:
I (788) heap_init: At 4081FB30 len 0005CAE0 (370 KiB): D/IRAM
I (795) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (801) heap_init: At 50000068 len 00003F80 (15 KiB): RTCRAM
I (809) spi_flash: detected chip: generic
I (812) spi_flash: flash io: dio
W (816) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (830) sleep: Configure to isolate all GPIO pins in sleep state
I (836) sleep: Enable automatic switching of GPIO sleep configuration
I (843) coexist: coex firmware version: b6d5e8c
I (849) coexist: coexist rom version 5b8dcfa
I (854) app_start: Starting scheduler on CPU0
[... 63 lines cut ...]
408229f0: 0x00000014 0x00000001 0x00000007 0x40822aec 0x40822aec 0x4087cff4 0x00000020 0x420d07ec
40822a10: 0x00009d40 0x00000020 0x00000000 0x40803d06 0x00000001 0x4087cff4 0x00000006 0x4204b8ce
40822a30: 0x40822b40 0x4087cff4 0x00000000 0x4204c2e2 0x4087ee70 0x4087cff4 0x00000006 0x4204cb56
40822a50: 0x000000ff 0x4087cff4 0x00000006 0x000000ff 0xff001401 0x0000000f 0x6574636f 0x00000074
40822a70: 0x00000000 0x00000000 0xffffffff 0xffffffff 0x4087ee70 0x4087ee70 0x00000000 0x4204be72
40822a90: 0x00000000 0x4087cff4 0x00000000 0x4204cac4 0x40822bfc 0x40822bfc 0x00000004 0x000000ff
40822ab0: 0xff004801 0x00000004 0x74726f70 0x5f70616d 0x00626174 0x00000000 0xffffffff 0xffffffff
40822ad0: 0x00000000 0x00000000 0x4082080c 0x00000000 0x00000000 0x00000001 0x40816fb8 0x00000074
40822af0: 0x00000001 0x420526ca 0x420526b4 0x42055b30 0x420527ec 0x42055b82 0x42055b66 0x42055bb8
40822b10: 0x42055b9c 0x420527fc 0x42055b18 0x42055d70 0x42056ac8 0x40000748 0x420526a6 0x42055bd2
40822b30: 0x42055bfa 0x42052698 0x42055c08 0x42055c30 0x42055c3e 0x42055ca8 0x42055e9e 0x4205f77e
40822b50: 0x4205f836 0x42052682 0x42055b00 0x42052686 0x0000000a 0x00000020 0x00000001 0x00000000
40822b70: 0x00000020 0x00000000 0x00000005 0x00000000 0x00000000 0x00000001 0x00000001 0x00000000
40822b90: 0x00000001 0x00000000 0x00000006 0x00000000 0x000002f0 0x00000020 0x00000001 0x00000000
40822bb0: 0x00000001 0x00000007 0x1f2f3f4f 0x42049240 0x408257ec 0x408257cc 0x0104a8c0 0x00ffffff
40822bd0: 0x0104a8c0 0x00000000 0x00000000 0x420495d0 0x00000000 0x421393e8 0x00001102 0x00000000
40822bf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4081f000 0x4200236c
40822c10: 0x00000000 0x00000000 0x40822c54 0x40822c3c 0x00000000 0x00000000 0x00000001 0x4080c3e2
40822c30: 0x00000000 0x4087ccf0 0x00000000 0x00000000 0x00000000 0x00000000 0x42128000 0x4080c3ea
40822c50: 0x00000000 0x00001388 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x4080c110
40822c70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40822c90: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x40822830
40822cb0: 0x00000007 0x4081a9c8 0x4081a9c8 0x40822cac 0x4081a9c0 0x00000018 0x4087de28 0x4087de28

ELF file SHA256: 7d3efeb99fd9872e

dchristl, 2024-03-03:
I've published a new release, containing the ESP32-C6 firmware. If you have any probleme, please let me know.

> I (1019) ESP32NRE: Setting the bandwith to 40 MHz
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42001d0a

Your error is only when connected to 802.11ax-network, because this is not supported. See [specifications (
Enhanced Connectivity)](https://www.espressif.com/en/products/socs/esp32-c6):

>ESP32-C6 supports a 20 MHz bandwidth for the 802.11ax mode and a 20/40 MHz bandwidth for the 802.11b/g/n mode.

The error should not occur anymore; however, the setting does not have an effect on ax networks.

closed 2024-03-03 by dchristl

ErfanDL, 2024-03-04:
> I've published a new release, containing the ESP32-C6 firmware. If you have any probleme, please let me know.
>
>
> Your error is only when connected to 802.11ax-network, because this is not supported. [...]

thank you. The only problem with the internal USB is that the factory reset pin is the same as the USB pin (GPIO12), which leads to a factory reset during startup.

dchristl, 2024-03-04:
> The only problem is, the factory reset pin is the same as the internal USB pin (GPIO12), which leads to a factory reset during startup. If possible, set the factory reset pin to another GPIO for esp32c6 build.

That's not usual, most other chips usually have it on pin 18 or 19. Could you please provide more details? Does this happen consistently, or only under certain conditions (e.g., after flashing)? Why doesn't @kondakos  have the issue? Are there different versions? 

Which pin on the C6 Dev Boards would be easily accessible or bridgeable for me to use?

ErfanDL, 2024-03-06:
> That's not usual, most other chips usually have it on pin 18 or 19. Could you please provide more details? Does this happen consistently, or only under certain conditions (e.g., after flashing)? Why [...]

this problem is consistently when using USB host (not UART for flashing). some esp32c6 dev boards dosn't have UART and using USB host pins for flashing (GPIO12 and GPIO13). if possible change the reset factory pin for esp32c6 to GPIO18.

dchristl, 2024-03-18:
@ErfanDL ,

could you please try it out, if everything works:
[firmware.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14634461/firmware.zip)

You can flash the firmware with:

`esptool.py  write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect  0x10000 firmware.bin`

ErfanDL, 2024-03-18:
> @ErfanDL ,
> 
> could you please try it out, if everything works: [firmware.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14634461/firmware.zip)
> 
> You can flash the firmware with:
> 
> `esptool.py write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 firmware.bin`
Hi.
Unfortunately, it has the problem of the previous version. Did you change the factory reset pin?

dchristl, 2024-03-18:
Yes, I changed it to GPIO 18. Maybe, I have picked the wrong firmware on upload. Can you please retry with 
[firmware.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14635231/firmware.zip)

Thank you

ErfanDL, 2024-03-18:
> Yes, I changed it to GPIO 18. Maybe, I have picked the wrong firmware on upload. Can you please retry with [firmware.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14635231/firmware.zip)
> 
> Thank you

thanks. the problem is solved with this firmware