dchristl/esp32_nat_router_extended #106: Request for ESP S3 chip

Kuba1606, 2024-03-04:
Hello, I am impressed with your work. This is exactly what I was looking for. Unfortunately, my module is based on the ESP32 S3 model (SpeedStudio Xiao S3), so your bin files for other models do not work. I'm not that advanced of a user to be able to compile my bin files for S3. Could you also compile your work for this chip? Thank you greetings.

8380861, 2024-03-09:
> 你好，你的工作给我留下了深刻的印象。这正是我一直在寻找的。不幸的是，我的模块基于 ESP32 S3 型号 （SpeedStudio Xiao S3），因此其他型号的 bin 文件不起作用。我不是能够为 S3 编译我的 bin 文件的用户。你能为这个芯片编译你的工作吗？谢谢你的问候。

您解决了在esp32s3上的使用此程序的问题了吗,如果解决可否给我提供一下解决方案呢

AlexNOliveira, 2024-03-14:
Hi @dchristl ,

   First of all, thank you for your work in this project. I have a S3 version, this one: https://www.aliexpress.com/item/1005006266375800.html
   
   I wrote the bin file you shared in your previous comment, but I didn't succeed to see the network the board should have created. I tried twice. 
   First time, I just did what is in the First install/ Reset section, including remove the board from the computer and connect it again. No network showed up.
   In the second time, I did the erase as suggested in the General section and repeated the process of First Install. No network showed up.

   In all cases there were no errors been shown in the console. I also tried the ESP Flash Tool, with no success.

AlexNOliveira, 2024-03-18:
Hi @dchristl ,

Apologies for my previous comments. Let me try to explain differently.

All my tests were done based on downloading the repo via a zip file.

My first attempt do upload a bin file to S3 was based on the zip file you shared in your comments to Kudo1606. Since it was (I assume in that way) a fresh install, I tried to upload the full file (esp32-s3-nat_extended_full.bin).

I tried to upload in 3 different ways, all following the readme file.

The first time, I followed the First Install process: **esptool.py write_flash 0x0 esp32nat_extended_full_vX.X.X.bin**
The second time, I followed the General process: **esptool.py erase_flash**, followed by the step above.
The third time, I followed the First Install for Windows process.

In all 3 cases above, no issues with uploading, I mean, there were no error messages after the upload processes.

It is important to highlight that I have no background in ESP32 nor in the development tools required to work with it. I am someone who just started to study it and try to solve the issues by google them. :-)

All these attempts resulted in not having the WiFi SSID "ESP32_NAT_Router" available for me, even removing and connecting the USB cable to power the S3.

I decided to open into PlatformIO and try to build by myself. I researched on the internet and found out that I need to add a "task" to platformio.ini to build to S3. So I added the following code:

**[env:esp32-s3]
platform = espressif32
board = esp32-s3-devkitm-1**

After I added this code, I clicked on the build button on the footer of the VS Code. A few moments after the build had started, an error showed up. After a search for the error message, I found the solution at https://github.com/martin-ger/esp32_nat_router/issues/48. The solution was to change a few configurations for the S3. To do so, I run the following command:
**platformio.exe run --target menuconfig --environment esp32-s3**

Changing the configuration per the mentioned URL, made it possible to build with no issues. After that, I choose to upload using PlatformIO.

After following these steps, I was able to connect to the WiFi SSID "ESP32_NAT_Router". I was able to open the configuration page. It worked this time because I just observed that on my Android phone, it was necessary to agree to keep connected to the unsecured network, which I didn't notice the last time I tried to connect.

kof698, 2024-03-19:
> Hello @Kuba1606 ,
> 
> Thank you for your kind words. The problem with adding new chips is not the compilation but rather the testing on my part. I don't have all the chips and variations, so it will be difficult. I have attached the firmware for you to try. You can test whether you can flash it and check if it works in general. Then I can add the S3 to the next release.
> 
> [esp32-s3-firmware.zip](https://github.com/dchristl/esp32_nat_router_extended/files/14481502/esp32-s3-firmware.zip)
> 
> Kind Regards, Danny

Hello developer, I use esp32s3 n16r8 development board flash size 16M after brushing into the firmware or can not identify the wireless route, the source code needs to modify where to adapt?

candrews67, 2024-05-19:
> Hello @Kuba1606 ,
>
> Thank you for your kind words. The problem with adding new chips is not the compilation but rather the testing on my part. I don't have all the chips and variations, so it will be [...]

this works for me!
ESP32S3

candrews67, 2024-05-19:
sptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:27:e2:1e:0a:e8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x0015afff...
Compressed 20928 bytes to 13290...
Wrote 20928 bytes (13290 compressed) at 0x00000000 in 0.4 seconds (effective 471.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 113...
Wrote 3072 bytes (113 compressed) at 0x00008000 in 0.0 seconds (effective 536.2 kbit/s)...
Hash of data verified.
Compressed 1354592 bytes to 757238...
Wrote 1354592 bytes (757238 compressed) at 0x00010000 in 8.6 seconds (effective 1261.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

dchristl, 2024-05-24:
Great, I will add it to the official releases when I have more time

reyxv16, 2024-06-21:
Hello, what is the possibility that you can compile a version for the esp32c3 super mini?  The current version for the C3 has not worked for me, I thank you very much, excellent work, greetings.

dchristl, 2024-06-22:
Hello @reyxv16 , 

If I understand the documentation correctly, the super mini version is also just a normal C3 on a different board. So the firmware should actually work (what does the og say when flashing, what does the serial console say). Unfortunately I can't support all chips in case it doesn't work.

kof698, 2024-10-13:
> Can you post an output of the serial connection after flash? The attached bin should work for all S3 models

[974/976] Linking CXX executable esp32_nat_router_extended.elf
FAILED: esp32_nat_router_extended.elf 
cmd.exe /C "cd . && D:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=D:/s1/build/esp32_nat_router_extended.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.wdt.ld -T esp32s3.rom.version.ld -T esp32s3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\esp32_nat_router_extended.elf.rsp -o esp32_nat_router_extended.elf  && cd ."
D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj):(.literal.main_task+0x24): undefined reference to `app_main'
D:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/freertos/libfreertos.a(app_startup.c.obj): in function `main_task':
D:/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:206:(.text.main_task+0xa8): undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

fiz1962, 2024-10-18:
I have a ESP32-S3 supermini and use Linux. I tried the version at martin-ger.  It seemed to flash ok but constantly rebooted (watched output in Arduino IDE Serial Monitor).  I rebuilt the version at martin-ger and it flashed but constantly rebooted.  So I found the version here compiled for a S3.  I erased the flash and then flashed the full version.  It seemed to flash ok but the serial monitor shows it constantly outputting "invalid header 0xffffffff".

fiz1962, 2024-10-18:
I went back to the martin-ger code, ran menuconfig and changed the flash from 16MB to 4MB, recompiled and flashed.  And now the martin-ger version works.  I suspect there is the same problem here.  The ESP32-S3 supermini only has 4MB flash.

funkypozzy, 2025-10-06:
> Hello [@Kuba1606](https://github.com/Kuba1606) ,
>
> Thank you for your kind words. The problem with adding new chips is not the compilation but rather the testing on my part. I don't have all the [...]

Thank you! This worked for my Seeed Studio XIAO ESP32S3 with settings as per attached image.

[image]

[image]