dchristl/esp32_nat_router_extended #108: Bandwidth Controller or  Load Balance

adi160885, 2024-03-11:
your application is very impressive, can you add a "Bandwidth Controller" to regulate *Mbps* or "Load balance", it might be very useful, thank you sir

dchristl, 2024-03-18:
Hello @adi160885 ,

this is sadly not possible, because the framework (and maybe the hardware, too) of the ESP32 doesn't support this.

closed 2024-03-18 by dchristl

adi160885, 2024-03-18:
What a shame, I use 4 ESP32 modules, all the average speed only reaches 3 -> 5 Mbps, is there something wrong with my ESP32 module, but if I use "gjroots ESP32-NAT Router +" the speed reaches 10 -> 14 Mbps. thank you in advance for answering my question 🙏

dchristl, 2024-03-18:
The routing code of this project and gjroot's one are based on martin-gers work, so there should be no difference.

adi160885, 2024-03-18:
Thank you for your answer, your project is very good