dchristl/esp32_nat_router_extended #109: Probleme de configuration du routeur

rdjules, 2024-03-13:
Bonjour à tous, je vous remercie déja pour la proposition, si j'ai bien compris le protocole, on doit se connecter via notre appareil, puis aller sur le site https://192.168.4.1. Malheureusement, je n'arrive pas jusqu'à cette partie là, une page saffiche m'annonçant que je n'ai pas accés à internet, si quelqu'un pourvait m'aider et m'envoyer un tutoriel video svp.
Cordialement

dchristl, 2024-03-18:
It seems your URL is wrong. It is http://192.168.4.1 (without the s), than it should work.

rdjules, 2024-03-18:
je viens de réeesayer avec le lien envoyé, cela ne marche pas non plus, il m'affiche juste la page qui indique que je n'ai plus de réseaux

gitfvb, 2024-03-18:
@rdjules Are you connected to the ESP32 network and which browser/device are you using?

rdjules, 2024-03-20:
Oui 

> @rdjulesÊtes-vous connecté au réseau ESP32 et quel navigateur/appareil utilisez-vous ?

Oui, je suis connecté en wifi à la carte, j'ai essayé de mon iPhone, avec safari, puis de mon xiaomi, puis de mon ordinateur , en utilisant des navigateur tel que chrome, firefox et même duckduckgo.
Si quelqu'un pouvait m'envoyer une vidéo de la manip à faire, cela m'aiderait svp.
Cordialement

gitfvb, 2024-03-20:
Just connect and open http://192.168.4.1 in your browser. You need to have DHCP activated in your device (which is the default).

rdjules, 2024-03-22:
Comment on active le DHCP ?

rdjules, 2024-03-22:
Comment je peux activer le DHCP ?

gitfvb, 2024-03-27:
To find this out, connect your client open a shell or cmd or PowerShell and enter `ipconfig`, this should give you the output if DHCP worked. The setting for DHCP depends on the clients operating system.