dchristl/esp32_nat_router_extended #110: Question about auth mode

gitfvb, 2024-03-16:
Hi @dchristl,

This project is really cool. Got it working so far and in a speedtest I get up to 15Mbit/s, which is very impressing, but when trying to connect my TV, it only supports WPA2PSK(AES), WPA2/WPA Mixed PSK and WPA3-Personal. Without encryption it works, but that was more for debug purposes.

I have found to right place to change the `.authmode` but I haven't found the possible options I can enter before compiling. And maybe `.pairwise_cipher` needs a change, too. Do you have any idea what to enter to get it working?

dchristl, 2024-03-18:
Hello @gitfvb ,

the current auth-mode already supports maximum compatibility. Normally, WPA2 PSK should also be possible. I wouldn't know what else I could do. If you want to compile and try it yourself, you can find possible values [here, starting at line 60](https://github.com/espressif/esp-idf/blob/release/v5.2/components/esp_wifi/include/esp_wifi_types.h). The cipher-possibilities start at line 178. 

Please let me know, what options work. 

Kind Regards,
Danny

gitfvb, 2024-03-18:
Nice @dchristl , thanks for that list. That was exactly what I was searching for. I suspected it to be the cipher, but CCMP was fine. Instead the TV needed the authmode `WIFI_AUTH_WPA2_PSK` instead of `WIFI_AUTH_WPA2_WPA3_PSK`

Now it works like a charm to stream Disney+, thanks :-)

closed 2024-03-18 by gitfvb

dchristl, 2024-03-20:
Great that it's working now, although I find it a bit strange. According to the documentation, WIFI_AUTH_WPA2_WPA3_PSK includes WIFI_AUTH_WPA2_PSK and automatically negotiates the possible encryption with the device. It seems like that's not working in your case. I'll look into whether I can expand on that or find out more.

gitfvb, 2024-03-20:
This setting definitely made the difference. I have also changed the `esp_nat_router.c` to this so it scans for all networks first and then connects to the one with the best rssi. In our house network we have multiple mesh devices and it can happen, that the device connects to the wrong repeater. I don't know if that change makes a difference yet.

```
    /* ESP WIFI CONFIG */
    wifi_config_t wifi_config = {
        .sta = {
            .scan_method = WIFI_ALL_CHANNEL_SCAN,
            .sort_method = WIFI_CONNECT_AP_BY_SIGNAL}};
```