dchristl/esp32_nat_router_extended #111: No results after wifi scanning ESP32-C6

kondakos, 2024-03-19:
After scanning wifi stations there is no results 
ESP32-C6 
 
https://github.com/dchristl/esp32_nat_router_extended/releases/download/v7.0.0/esp32c6nat_extended_full_v7.0.0.zip

dchristl, 2024-03-20:
Can you scan on a different location? If you never see any SSID, your device may be defective. Does the router work, when you input your connection data without scanning before?

kondakos, 2024-03-20:
You are right mine fault. Scan and results works excellent. The issue closed.

closed 2024-03-20 by kondakos

AILIFE4798, 2024-03-25:
@kondakos can you tell me whats the performance on the c6
is it any better then the og esp32
the c6 have wifi6 but still 2.4g and also the cpu clock is lower on c6 so im curious is it worth replacing

kondakos, 2024-03-26:
Good results . Direct connection from 4m to home WIFI 2.4 Mhz AP from 21 Mbps/4Mbps goes to 7Mbps/3Mbps when connection from 4m with the ESP32C6. The half bandwith is normal as all repeaters with one 2.4ghz chip. Half the time acts as receiver (cleint) and half as station (AP). 
I prefer Seeeduino Seeed Studio XIAO ESP32-C3   (U.FL)   with  extra wifi antenna https://www.aliexpress.com/item/1005001522213744.html   with excellent home coverage (long distance 100 open space).

AILIFE4798, 2024-03-26:
@kondakos the original esp32 can do about 15mbps max
i just did a test with 1 wall in between the main router and esp and got 11/8M
the esp32 c series is the low end series so i wonder if its any good, especially the esp32c6 cost more then $10 and the original esp32 only cost $2
because your original ap can only do so little anyways its not a very good test
but it at least proves its better then the esp8266 which i have to use for 2 year to have any internet in my home which only can get about 4mbps max and 1~2.5mbps in reality as the signal is very weak or i wouldnt need a repeater
now i have internet at my home now no longer need esp repeater to provide my own use but i want to use it for iot device that is far away and keep disconnecting from home assistant
when the esp32c5 come we will finally have indivisual radio one as ap and one as station so hopefully seeing 50m or more

kondakos, 2024-03-26:
@AILIFE4798. As you write you will connect router->esp32c5 (as AP) ->    esp32c5 (as client) so the best throughout will be  50% of the router 2.4 Ghz Wifi.
My tests is not the best (but can be usefull to compare your setup) as my openwrt router 2.4 Ghz Wifi  is optimized for more distance  clients.   
The Seeeduino Seeed Studio XIAO ESP32-C3 (U.FL)  https://www.aliexpress.com/item/1005006353129383.html with extra wifi antenna https://www.aliexpress.com/item/1005001522213744.html is  an option.

AILIFE4798, 2024-03-26:
the c5 will have 2.4g and 5g so it can do more then 50% bandwidth
but it is not sold yet so gotta use either the c3 or c6 for now
the c3 is a ultra low cost and low end chip designed to replace esp8266 so i expect it to perform the same
my question is esp32c6 have wifi6, is it any good?or should i just choose the c3 thats 10 times cheaper?or does all the c series sucks and i should use the og esp32

AILIFE4798, 2024-03-26:
@kondakos

kondakos, 2024-03-27:
@AILIFE4798 My experience with c3 and c6 is that  both can handle light wifi traffic like radio streaming  (3<Mbps)  at far  50-100m. Heavy trafic wifi like file transfer (copy and paste)  can be problematic.

AILIFE4798, 2024-03-27:
ok, then ill get a esp32c3 and test it out