dchristl/esp32_nat_router_extended #112: signal strength : 0db

1275cubmanestate, 2024-03-19:
hi, i'm french guy and a recently buy an esp32 i have flash with the last realese 7.0.0 and my signal is 0 db with my actual router so i can't use my esp like wifi extender.
[image]
when i connect it with my iphone acces point it works

any idea ? 

my internet router is 
[image]

dchristl, 2024-03-20:
Hello, 

do you see in your upstream router whether the ESP has connected? Have you tried getting closer or using a different network (e.g., a mobile hotspot)?

ikkair, 2024-04-23:
Hello, I too have 0db issue with my ESP32.

Im using ESP32-D0WD-V3 chip, since the version 7.0.0 The scan result is nonexistent and the signal is always 0db. My ESP32 variation is using external antennae but i have checked it is working properly in version 6.5.0. The board is ESP-32 DEVKITC V4 WROOM-32U.

dchristl, 2024-04-26:
It seems there is a bug in the newest esp-idf-version.   I will revert the changes and release a hotfix soon. Could you please check if it will work again? Thi s only affects scanning. The signal works like expected. 

Thank you.

ikkair, 2024-04-29:
Thanks for the fix, 

It fixed the scan, connection 0db, and custom third octet issue.
My device now worked flawlessly, the remaining problem is just the long ssid (or special character) issue, the scan can detect the ssid and when i connect then i fill the password, it just have 0db. 
the special character is underscore "_" inside the ssid (or password), when it is set and the ssid length is more than 15 it just wont connect for me. 

But it is not a big problem, I just rename the ssid and change the password to fix the issue