dchristl/esp32_nat_router_extended #113: Neopixel RGB

ErfanDL, 2024-03-20:
Hello. It would be great if you add RGB neopixel feature because some boards only have neopixel RGB. also possible to select the GPIO specific to the neopixel pin in the settings.

thanks.

dchristl, 2024-03-20:
Hello, 
I'm sorry, but I can't really achieve that. There are numerous boards with different configurations, and it would be a nightmare to support them all. I prefer to focus on the core functionality and expand on that. Personally, I don't see much value in having lighting on a router. However, feel free to branch and expand it if you'd like.

AILIFE4798, 2024-03-25:
what is the default gpio for normal led 
and yes because every board use different gpio can we configure it in settings which gpio to use

AILIFE4798, 2024-03-28:
my dev board have normal led on gpio 22 and ofcourse it didnt work
idk where can i set it

dchristl, 2024-03-28:
> idk where can i set it

As I mentioned before, there is no option for that, and even if there were one, the Neopixels wouldn't work because they are controlled differently than a regular LED.

RobKeij, 2024-11-07:
[image]

Led iis not working because connect_count is always 0
[image]