dchristl/esp32_nat_router_extended #115: Static Ip address for the ESP32

DadaneMC, 2024-04-04:
All the options work well, but how can I assign a static IP address to the ESP32 if my router simultaneously has MAC address filtering and IP address filtering?

dchristl, 2024-04-05:
Sorry, I don't quite understand the question. There isn't a setting to assign a static IP to the router, and it's usually unnecessary. Any standard router should be able to assign the same IP to a device consistently

DadaneMC, 2024-04-08:
My question wasn't very clear, sorry. I wanted to ask how to assign a static IP to my ESP32 when it connects to my router. For example, the gateway of my router is 192.168.1.1, and I want my ESP to have the static IP address 192.168.1.3 when it connects to my WiFi (router). This is because my router doesn't follow the DHCP protocol.

dchristl, 2024-04-12:
What do you mean by your router doesn't follow the DHCP protocol? Do you have to assign a static IP address to every device that connects to the router? How do you handle that for IoT devices, that doesn't have such settings? That sounds very strange. What model of router do you have? Even the cheapest routers should be able to handle that. 
With this project, unfortunately, it's currently not possible, and implementing such functionality would also require some development effort. At the moment, I don't have the time for it, and I have other more pressing matters on my agenda.

DadaneMC, 2024-04-13:
I use static IP addressing for added security. I hope this feature is feasible. I completely understand that you're busy at the moment. Thank you for taking the time to respond to me.

dchristl, 2024-04-13:
Static IPs don't actually contribute to security; they just ensure that devices are reachable at the same address. This is typically managed by the DHCP server, both at home and in enterprise environments. If you prefer to assign static IPs, simply increase the DHCP lease time for the device and assign a fixed IP. This way, no other device can take that IP. It's all done on the router and can be made more secure by restricting access based on MAC address, for example.

NaxGH, 2025-04-05:
The original code is always there, I try to migrate from [martin-ger's ESP32 NAT Router-project](https://github.com/martin-ger/esp32_nat_router) and test using hotspot and I wonder why I dont have Internet and I don't see my ESP32 on my phone, turn out I have set static IP before and still save in NVS.
Anyway, I want to point out some router will assign wrong IP address for some reason. Example in my case the router will assign 192.168.1.10 but it should be 192.168.80.10