dchristl/esp32_nat_router_extended #116: Adding bluetooth proxy

stalniy, 2024-04-10:
Would be nice to additionally have an option to extend bluetooth range.

## My usecase

I want to use this repeater outside home, to have a good wifi connection near my gates, so when I arrive home my phone can quickly connect and then I can open garage gate using homeassistant. 

Additionally I have 1 bluetooth sensor on my water accumulator (which is also outside home) and it would be nice to proxy its BLE signal to WIFI router (or MQTT).

dchristl, 2024-04-12:
Hello @stalniy ,

I don't quite see your use case fitting into this project's scope. It seems like you need something more standalone because it doesn't directly align with the project's goals. You would need some kind of translation layer between Bluetooth and WiFi. I'm also having trouble envisioning how it would work in general. You can't simply translate Bluetooth to HTTP/MQTT; at most, you could capture the payload and send it somewhere. There might be simpler solutions, such as using a sensor that natively supports WiFi and MQTT. If the sensor is outside your WiFi range, you could use this project to connect it to WiFi.

Regards,
Danny

stalniy, 2024-04-13:
Ok thanks for explanation. 

In most cases, sensors that can measure air pressure inside close space (mostly for wheel tires) are using either RF or bluetooth. I can’t find WI-FI or zigbee based sensor for this

Expansion tank is installed outside home, that’s why I need Bluetooth proxy to extend signal from Bluetooth sensor

closed 2024-04-13 by stalniy